PartialVolume/shredos.x86_64 #306: ShredOS consistently freezes partway through DoD Short

internet-tubes, 2024-12-19:
I'm trying to wipe a 4TB physical disk, and every time I try to run a DoD Short wipe it freezes between 2-6 hours, completely locking up the machine.

I'm using a USB drive (created with Rufus) of the latest ShredOS image. I tried adding nomodeset to the config files, which were required in order for it to boot at all (RTX 3080), but it freezes.

Specs:
Drive to be wiped: 3.5" 4TB WD Black
GigaByte Aorus Master Z390
Intel 9900K
Nvidia RTX 3080 10GB
I will be using Samsung's Secure Erase tool for the SSDs

PartialVolume, 2024-12-20:
What's the background history of this system? Have you been using it yourself without any issues or is it an unknown.

Have you run overnight memtest diagnostics on it to verify the CPU memory? Alternatively, if it's got multiple SIMMS try running it with one then the other to see if one of the SIMMS has a fault.

internet-tubes, 2024-12-20:
It's my gaming rig, zero known issues. I'm just wiping the storage HDD before I sell it.

PartialVolume, 2024-12-20:
When it freezes does ALT F2 i.e switching between terminals still work?

 Although percentage completion may have stopped what's ETA (top right of display) doing? Is it still counting?

Instead of a more time consuming wipe like DoD 3, does a quick wipe like zeros, no verification, no blanking work? (single pass of zeros)

internet-tubes, 2024-12-20:
I setup a USB with the previous version as a test last night (DoD Short again), and when I checked this morning it froze at 5:33:xx. The timer stopped, and Alt F2 was not responsive. 

I have not tried a shorter method yet. 

Would a different USB stick matter? I noticed the stick is warm when I went to power off the machine.

PartialVolume, 2024-12-20:
The usb stick is only used to load the ShredOS.img, once Linux has started booting the usb is not used until you hit enter when the wipe completes. The usb is not continuously mounted like other operating systems. It is only mounted at boot time and for writing the certificate on completion. Once nwipe is running you can pull the stick out as it's not required during wiping. Even if you forgot to plug the stick back in as long as you didn't reboot or shutdown but selected the option to restart nwipe then control C to exit nwipe it would still attempt to write any certificates that hadn't been written back to the stick.

Personally I'd still run memtest to verify the CPU memory. As it's a gaming rig I'm assuming all fans are running, heatsinks not clogged up and your not over clocking which could cause very random issues. I suppose it's possible the drive itself could be hanging the OS but if you have other drives you could test that.

Like I said, do a simple zeros wipe, no verification, no blanking which will be relatively quick on a 4TB drive  to see if it passes  or not. If it works do a prng stream. Starting with a simple wipe first might help in identifying what the issue is. A method using a prng stream is more likely to trigger a memory error if CPU memory is at fault, that why it's better to start simple in order to identify a pattern which may lead us to the cause of the problem.

Firminator, 2024-12-20:
Or take out the troublesome drive and hook it up to another PC, then check out the SMART status of drive to rule out any issues ( https://gsmartcontrol.shaduri.dev/downloads ).
If the drive checks out good, then ShredOS it from that PC and see if the wipe finishes.

internet-tubes, 2024-12-27:
I ended up getting it to work first try (with DoD short) in another machine. I have no idea what was causing this, as these are the things I tried on my current machine and failed each time between 3-6 hours:

- Moved RAM to another slot
- Got different RAM sticks altogether
- Removed the GPU and plugged HDMI into integrated graphics
- Enabled CSM and legacy settings
- Reset CMOS / BIOS settings

It's wiped now, but I'm curious what the freezing issue stemmed from.

closed 2024-12-27 by internet-tubes