PartialVolume/shredos.x86_64 #308: Cannot allocate kernel

Torwass, 2024-12-27:
Hello,

I can't boot ShredOS from USB flash drive. After Grub and ShredOS have finished loading, I get message "error: cannot allocate kernel" followed by "Failed to boot both default and fallback entries". Then, I'm back to the Grub choose menu, where I can choose to boot ShredOS again or switch to Command Line Mode, or edit parameters. Choosing to boot again displays only first message "error: cannot allocate kernel", then return me to the choose menu.

I tried to burn both latest and previous versions of ShredOS with Rufus on 2 different USB drives, but that didn't help.

My BIOS is in UEFI mode and Secure Boot is disabled. 
Motherboard: Gigabyte Z490 Vision D
CPU: i7-10700K
GPU: GeForce 1050Ti

PartialVolume, 2024-12-27:
You may get that error when there is some issue with the CPU memory. If you can, try confirming the usb stick boots on another system. I would also run memtest or if using multiple SIMMS try booting with each one in turn assuming that will work on that motherboard.

Torwass, 2024-12-31:
Thanks for suggestions! I've tried them, but it didn't work out. It turned out today, the problem was the way I was trying to boot. I did it through the "F12 Choose Boot" option. Changing boot order in BIOS instead solved the problem.

closed 2024-12-31 by Torwass

PartialVolume, 2025-01-01:
Glad you sorted it, strange that boot order would create an error like "cannot allocate kernel", however something to keep in mind for future reference.