PartialVolume/shredos.x86_64 #311: extend image instead of using gparted

boruto79, 2025-01-05:
I was able to make a kernel 6.12.8 using a tarball. when trying to do it with a updated buildroot it keep on having issue. Question 1. is there a way to make the image bigger instead of using gparted so i have room for the pdf. 2: Is there a way to use the nvme-cl in nwipe instead of Alt-F2?

PartialVolume, 2025-01-05:
1. Edit ../board/shredos/genimage.cfg, find the line `        size = 280000000` that's the size in bytes. To double the size change it to `size = 560000000`
2. No, at least not yet, but nvme erasure using the drives firmware is in the list of features to be added.

boruto79, 2025-01-06:
Okey thanks, Another question can you add an option to read the firmware and add make, model, asset tag and serial number of the computer. will help a lot to keep track of the pdfs.

PartialVolume, 2025-01-06:
>can you add an option to read the firmware and add make, model, asset tag and serial number of the computer. will help a lot to keep track of the pdfs.

Yes, that's in progress, it will be a host specific PDF report showing host details and all the drives on the host that were wiped. Nwipe will produce both types of certificates.