PartialVolume/shredos.x86_64 #316: Switch from telnet to SSH

fthobe, 2025-01-16:
Hey, 

I think it's time to switch from telnet to SSH. Is this something I can try to tackle this weekend?

[Systemrescue has a very valid approach in my opinion](https://www.system-rescue.org/manual/Network_configuration_and_programs/).

PartialVolume, 2025-01-16:
Not switch, telnet has it's users in certain network configurations when the user is well aware of the security implications. Despite the fact telnetd is disabled by default. ssh would be in addition to the telnet option.

fthobe, 2025-01-16:
ok :)

Out of curiosity, which are the reasons to use telnet over SSH?

PartialVolume, 2025-01-16:
Simplicity, it doesn't require any configuration. On a LAN with just two hosts and no WAN connection or a properly configured firewall you don't need ssh. I use ssh all the time over the internet for admin and agree ShredOS should have ssh but I wouldn't remove the telnet option as there are people like me that know when you should use it and when you absolutely shouldn't use it. But ShredOS should have ssh for those that want to login over a insecure LAN or over the internet.

Firminator, 2025-01-17:
Yes we had that discussion before. You brought up the same use case which is valid.
I don't even have my wiping device wired at all :) but that's probably an exception.