PartialVolume/shredos.x86_64 #320: Cannot build current commit due to dead host-tar package

Knogle, 2025-01-22:
Hey, i hope you are doing fine.
It seems like, mirrors for host-tar are dead or something. Does it work for you?

```
>>> host-tar 1.35 Downloading
wget -nd -t 3 -O '/tmp/shredos.x86_64/output/build/.tar-1.35.cpio.gz.iHMr7g/output' 'http://ftpmirror.gnu.org/tar/tar-1.35.cpio.gz' 
--2025-01-22 19:50:46--  http://ftpmirror.gnu.org/tar/tar-1.35.cpio.gz
Resolving ftpmirror.gnu.org (ftpmirror.gnu.org)... 2001:470:142:5::200, 209.51.188.200
Connecting to ftpmirror.gnu.org (ftpmirror.gnu.org)|2001:470:142:5::200|:80... connected.
HTTP request sent, awaiting response... 302 Moved Temporarily
Location: http://mirror.us-midwest-1.nexcess.net/gnu/tar/tar-1.35.cpio.gz [following]
--2025-01-22 19:50:47--  http://mirror.us-midwest-1.nexcess.net/gnu/tar/tar-1.35.cpio.gz
Resolving mirror.us-midwest-1.nexcess.net (mirror.us-midwest-1.nexcess.net)... 2607:fad0:3606::1:1361:3303, 165.84.218.160, 185.145.13.115
Connecting to mirror.us-midwest-1.nexcess.net (mirror.us-midwest-1.nexcess.net)|2607:fad0:3606::1:1361:3303|:80... failed: Connection timed out.
Connecting to mirror.us-midwest-1.nexcess.net (mirror.us-midwest-1.nexcess.net)|165.84.218.160|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 4625633 (4.4M) [text/plain]
Saving to: ‘/tmp/shredos.x86_64/output/build/.tar-1.35.cpio.gz.iHMr7g/output’

/tmp/shredos.x86_64/output/build/.tar-1.35.cpio.gz.iHMr7g/outpu 100%[=====================================================================================================================================================>]   4.41M  4.27MB/s    in 1.0s    

2025-01-22 19:52:58 (4.27 MB/s) - ‘/tmp/shredos.x86_64/output/build/.tar-1.35.cpio.gz.iHMr7g/output’ saved [4625633/4625633]

tar-1.35.cpio.gz: OK (sha256: c77a38fcf25b21fd8209d20d35638744344ded239cfc7df80138bf46d3c6b16d)
>>> host-tar 1.35 Extracting
mkdir -p /tmp/shredos.x86_64/output/build/host-tar-1.35
cd /tmp/shredos.x86_64/output/build/host-tar-1.35 && gzip -d -c /tmp/shredos.x86_64/dl/tar/tar-1.35.cpio.gz | cpio -i --preserve-modification-time
39210 blocks
mv /tmp/shredos.x86_64/output/build/host-tar-1.35/tar-1.35/* /tmp/shredos.x86_64/output/build/host-tar-1.35
rmdir /tmp/shredos.x86_64/output/build/host-tar-1.35/tar-1.35
>>> host-tar 1.35 Patching

Applying 0001-lib-getrandom.c-fix-build-with-uclibc-1.0.35.patch using patch: 
patching file gnu/getrandom.c

Applying 0002-Fix-boundary-checking-in-base-256-decoder.patch using patch: 
patching file src/list.c
Reversed (or previously applied) patch detected!  Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file src/list.c.rej
make: *** [package/pkg-generic.mk:247: /tmp/shredos.x86_64/output/build/host-tar-1.35/.stamp_patched] Error 1
```

PartialVolume, 2025-01-22:
I think it worked when I tried it about a week ago. Let me try a build and see what the problem is.

I also see some patch issues there. I think I know what that might be. Give me a couple of hours or so and I'll get back to you.

PartialVolume, 2025-01-22:
I'm not seeing any /output/build/host-tar-1.35 in my build directory. I'm building the latest master. Here's the host- software in my build. What version are you building?

``` c
host-acl-2.3.2                                   host-grub2-2.12                                                   host-python-calver-2022.6.26
host-attr-2.5.2                                  host-kmod-31                                                      host-python-flit-core-3.10.1
host-autoconf-2.72                               host-libconfuse-3.3                                               host-python-glslang-11.13.0
host-autoconf-archive-2023.02.20                 host-libffi-3.4.6                                                 host-python-hatchling-1.25.0
host-automake-1.16.5                             host-libglib2-2.76.1                                              host-python-hatch-vcs-0.4.0
host-binutils-2.42                               host-libopenssl-3.4.0                                             host-python-installer-0.7.0
host-bison-3.8.2                                 host-libtool-2.4.6                                                host-python-mako-1.3.6
host-bzip2-1.0.8                                 host-libxml2-2.13.4                                               host-python-markupsafe-3.0.2
host-dbus-1.14.10                                host-libzlib-1.3.1                                                host-python-packaging-24.2
host-dbus-glib-0.112                             host-llvm-15.0.3                                                  host-python-pathspec-0.12.1
host-e2fsprogs-1.47.1                            host-llvm-cmake-15.0.3                                            host-python-pip-24.3.1
host-expat-2.6.4                                 host-localedef-2.40-18-g5641780762723156b0d20a0b9f7df1d76831bab0  host-python-pluggy-1.5.0
host-fakeroot-1.36                               host-lzma-4.32.7                                                  host-python-poetry-core-1.9.1
host-flex-2.6.4                                  host-m4-1.4.19                                                    host-python-pypa-build-1.2.2
host-gawk-5.3.1                                  host-makedevs                                                     host-python-pyproject-hooks-1.2.0
host-gcc-final-13.3.0                            host-meson-1.6.0                                                  host-python-setuptools-75.3.0
host-gcc-initial-13.3.0                          host-mkpasswd                                                     host-python-setuptools-scm-8.1.0
host-genimage-18                                 host-mpc-1.3.1                                                    host-python-trove-classifiers-2024.10.21.16
host-gettext                                     host-mpfr-4.1.1                                                   host-python-typing-extensions-4.12.2
host-gettext-gnu-0.22.4                          host-nasm-2.16.03                                                 host-python-wheel-0.45.0
host-gmp-6.3.0                                   host-ncurses-6.4-20230603                                         host-skeleton
host-go-1.23.2                                   host-ninja-1.11.1.g95dee.kitware.jobserver-1                      host-swig-4.1.1
host-go-bootstrap-stage1-1.4-bootstrap-20171003  host-openssl                                                      host-tzdata-2024a
host-go-bootstrap-stage2-1.19.13                 host-patchelf-0.13                                                host-util-linux-2.40.2
host-go-bootstrap-stage3-1.21.8                  host-pcre2-10.43                                                  host-zic-2024a
host-go-src                                      host-pkgconf-2.3.0                                                host-zlib
host-gperf-3.1                                   host-python3-3.12.7
```

PartialVolume, 2025-01-22:
Not only don't I have any host-tar in packages but the tar package I do have has no patch files in the latest buildroot.

Have you got any old patch files from a earlier version left in a package directory? `Reversed (or previously applied) patch detected!` is the error you get when you've got old patch files from a previous version of buildroot left in the package folder of a particular program.

```
Applying 0002-Fix-boundary-checking-in-base-256-decoder.patch using patch: 
patching file src/list.c
Reversed (or previously applied) patch detected!  Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file src/list.c.rej
make: *** [package/pkg-generic.mk:247: /tmp/shredos.x86_64/output/build/host-tar-1.35/.stamp_patched] Error 1
```

PartialVolume, 2025-01-23:
Ok, I think I found the problem, the github version of ShredOS has the old patch files, unlike my local copy. I'll figure out why that is and update the Github master ASAP and let you know when it's ready to download.

PartialVolume, 2025-01-23:
@Knogle I've pushed commit c6c3355 which removes old patch files left over from the previous buildroot and would have caused compile problems just like the example you experienced above.

Feel free to clone the master again. I'm going to do a clean build overnight and make sure there are no further issues. I'll update you sometime tomorrow.

Knogle, 2025-01-23:
Thanks a lot :) Now failing at a different stage.
Old releases are building fine.

```
Emit Tests for uevent
Emit Tests for user
Emit Tests for user_events
Emit Tests for vDSO
Skipping non-existent dir: mm
Emit Tests for x86
Emit Tests for zram
make[1]: Leaving directory '/tmp/shredos.x86_64/output/build/linux-6.11.11/tools/testing/selftests'
ERROR: architecture for "/usr/lib/kselftests/tc-testing/action-ebpf" is "Linux BPF", should be "Advanced Micro Devices X86-64"
make: *** [package/pkg-generic.mk:402: /tmp/shredos.x86_64/output/build/linux-tools/.stamp_installed] Error 1
```

PartialVolume, 2025-01-23:
Yes, I've got another fix to make to the go package but the problem you describe above is a bug in 5.11. If you just run make again it then works ok.

I'll commit another fix in an hour.

PartialVolume, 2025-01-23:
go package fix commited 4fcdcef

It should compile ok now except for the BPF error

```
make[1]: Leaving directory '/tmp/shredos.x86_64/output/build/linux-6.11.11/tools/testing/selftests'
ERROR: architecture for "/usr/lib/kselftests/tc-testing/action-ebpf" is "Linux BPF", should be "Advanced Micro Devices X86-64"
make: *** [package/pkg-generic.mk:402: /tmp/shredos.x86_64/output/build/linux-tools/.stamp_installed] Error 1
```

When I get that error, I just run make again it will carry on where it left off. This BPF error seems to be a thing if you google

```[ERROR: architecture for /usr/lib/kselftests/tc-testing/action-ebpf is Linux BPF , should be Advanced Micro Devices X86-64](https://www.google.com/search?```

It's not obvious what the solution is, other than to upgrade to the next kernel.

As all we need to do is run `make` after getting this error it doesn't cause us much of a problem, ShredOS still builds ok.

Personally I don't want to spend much time on it but if you fancy looking in to it, please feel free.

PartialVolume, 2025-01-23:
@Knogle I'm just drafting the `2024.11_27_x86-64_0.38` release which I'm planning on publishing in the next day or two. I just want to check you build the current master ok, wrote it to USB and booted ShredOS 2024.11_27_x86-64_0.38?

Knogle, 2025-01-24:
Ahoy! Yep, after running make twice, i was able to successfully build the image!

closed 2025-01-25 by Knogle