PartialVolume/shredos.x86_64 #327: Booting with Z790 Asus Motherboard with 64G of memory not working

meaganmargaret, 2025-02-04:
Can't boot as the system used can seem to read the memory in my Z790 motherboard.  I had this issue with an older version of Linux Mint, but newer versions (v22) can read my memory just fine.  Any workarounds for this?  Some setting somewhere?

PartialVolume, 2025-02-04:
What's the error message?

meaganmargaret, 2025-02-04:
"error:  out of memory...."  is all I get.....

I then tried the Ventoy method mentioned in your Github post and I got the same result - even with the Ventoy ISO method.  Grub2 does not work and the first listing doesn't work.  With the Ventoy  method I got  a bit farther, I did see a menu show up, but none of the options (there were three) would work.

This is very similar to when I tried to upgrade from Linux Mint 20 to 22.  Basically I couldn't, and had to boot from v22 and reinstall everything.  But then it worked.

PartialVolume, 2025-02-04:
Do you have XMP enabled?

meaganmargaret, 2025-02-04:
Yes

PartialVolume, 2025-02-04:
I would temporarily disable XMP and see if it then works. I could be overclocking the memory is making it intermittently unstable depending upon your RAM type.

meaganmargaret, 2025-02-04:
So, I turned off XMP and still got the same error.

PartialVolume, 2025-02-04:
Have you run memtest86 to check for bad memory?

Firminator, 2025-02-04:
Sounds like grub is running out of memory due.
Adding GRUB_GFXMODE=800x600 is supposed to fix it (that would be needed to be added to the ShredOS grub commandline; see how @ https://github.com/PartialVolume/shredos.x86_64?tab=readme-ov-file#how-to-edit-the-shredos-efibootgrubcfg-and-bootgrubgrubcfg-files-when-using-ventoy-with-shredos-img-files )
Do you have discrete video or use the GPU in CPU?

meaganmargaret, 2025-02-04:
Oh God….yes!  This is a perfectly functioning PC which runs Linux Mint v22 (perfectly), Windows 11, and MacOS Monterey.  All work just fine, with no issues with the existing memory and all have been for almost a year.

meaganmargaret, 2025-02-05:
I have a discrete GPU, but after looking at your post, this is way over the top.  I’ll pass.  I just wanted to see if it would work for me.

I think your kernel is too old,   I have no problems with newer.

I hope you get this worked out, but for now, I’m out.

PartialVolume, 2025-02-05:
There may be insufficient memory because you have a large amount of memory being reserved in your bios settings for the video card. This may be reducing the amount of space in the first 4GB that's used to load ShredOS. ShredOS probably occupies more space than those other OS, due to its virtual RAM disc that contains  the entire OS.

So if there is a setting in the bios for adjusting the amount of memory allocated to graphics, then reducing it may fix the problem.

I've not seen this on any of my test systems, even the gaming laptop but maybe you have a high end GPU.

PartialVolume, 2025-02-05:
https://community.intel.com/t5/Blogs/Tech-Innovation/Client/Intel-Core-13th-and-14th-Gen-Desktop-Instability-Root-Cause/post/1633446#M40

Interesting, I've never known a CPU degrade over time. Apparently one of the symptoms can be out of memory errors. It apparently occurs under elevated voltage and temperature settings.

Anyway, as the op isn't interested in looking into this further, I'm closing this as 'out of memory' seems to crop up quite a lot on these motherboards for all operating systems.

closed 2025-02-05 by PartialVolume

PartialVolume, 2025-02-05:
This makes quite a read https://www.pcgamer.com/hardware/processors/intel-cpu-crashes-what-you-need-to-knowmicrocode-to-blame-but-fix-incoming-this-month-alongside-two-year-extended-warranty/

There seems to be upteen different ways the processor is failing, under load and at idle due to incorrect voltage requests and  certain motherboard manufacturers that don't stick to lower voltages 1.4v. Apparently not just gaming rigs but some servers are also seeing this problem.

I think I'll be sticking to my 12th gen Intel for a few more years. If there's one thing I hate, it's flaky unreliable hardware.

PartialVolume, 2025-02-05:
@meaganmargaret One last question, as I'm still curious about this problem, are you running one of these processors?

[image]

GalaxyM105, 2025-06-02:
I had this same message on a Dell Precision 7530 windows laptop and I couldn't find a place in bios to free up the internal Graphic Card memory, but I tried setting my boot mode to legacy boot and then booted into Ventoy and then into ShredOS and it let me into nwipe. I'm going to run the wipe now to see if it completes and I'll let you know.