PartialVolume/shredos.x86_64 #328: Resolution issues 2K Monitor and 4K TV

BestNoob, 2025-02-05:
U used shredos-2024.11_27_x86-64_0.38_20250123
on my USB stick created with Universal USB Installer - UUI

Now on my Monitor 27" but also my TV 43" everything is displayed so tiny i cannot see anything.
Its just a BlueScreen with white letters. DBAN is displayed fine.

Hope you can fix this.

PartialVolume, 2025-02-05:
Does pressing the f key change the size?

You will probably find that the nomodeset version will display with larger text much like dban.

Have never tried plugging it into my 60" TV, I'll try that out later.

PartialVolume, 2025-02-05:
There are a couple of ways this could be done:

The first is to use xrandr, however this isn't included in ShredOS by default. xrandr requires the dependency xorg7 so I would need to look at how much bloat gets added if I was to enable that.

The second is to use the kernel command `video=HDMI-A-1:1280x720@75` or something compatible with your hardware.

I guess you could hook up your normal linux distro to your TV and type `xrandr` which would show you all the available resolutions as shown below on my TV, the primary screen being a laptop.  You could then choose the resolution you want to use, using the data below I might go for `video=HDMI-1-0:1280x1024@75.02`
```
xrandr
Screen 0: minimum 320 x 200, current 5760 x 2160, maximum 16384 x 16384
eDP-1 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 382mm x 215mm
   1920x1080    144.02*+  60.01   144.02  
   1680x1050    144.02  
   1400x1050    144.02  
   1600x900     144.02  
   1280x1024    144.02  
   1400x900     144.02  
   1280x960     144.02  
   1440x810     144.02  
   1368x768     144.02  
   1280x800     144.02  
   1152x864     144.02  
   1280x720     144.02  
   1024x768     144.02  
   1024x768i    144.02  
   960x720      144.02  
   928x696      144.02  
   896x672      144.02  
   1024x576     144.02  
   960x600      144.02  
   832x624      144.02  
   960x540      144.02  
   800x600      144.02  
   840x525      144.02  
   864x486      144.02  
   700x525      144.02  
   800x450      144.02  
   640x512      144.02  
   700x450      144.02  
   640x480      144.02  
   720x405      144.02  
   720x400      144.02  
   684x384      144.02  
   640x400      144.02  
   576x432      144.02  
   640x360      144.02  
   640x350      144.02  
   512x384      144.02  
   512x384i     144.02  
   512x288      144.02  
   416x312      144.02  
   480x270      144.02  
   400x300      144.02  
   432x243      144.02  
   320x240      144.02  
   360x202      144.02  
   360x200      144.02  
   320x200      144.01  
   320x180      144.02  
   320x175      144.01  
DP-1 disconnected (normal left inverted right x axis y axis)
DP-2 disconnected (normal left inverted right x axis y axis)
DP-3 disconnected (normal left inverted right x axis y axis)
DP-4 disconnected (normal left inverted right x axis y axis)
HDMI-1-0 connected 3840x2160+1920+0 (normal left inverted right x axis y axis) 1872mm x 1053mm
   3840x2160     30.00*+  59.94    50.00    29.97    25.00    23.98  
   4096x2160     59.94    50.00    29.97    25.00    24.00    23.98  
   1920x1080     60.00    59.94    50.00    29.97    25.00    23.98  
   1680x1050     59.95  
   1600x900      60.00  
   1440x900      59.89  
   1366x768      59.79  
   1280x1024     75.02    60.02  
   1280x800      59.81  
   1280x720      60.00    59.94    50.00  
   1152x864      75.00  
   1024x768      75.03    70.07    60.00  
   800x600       75.00    72.19    60.32  
   720x576       50.00  
   720x480       59.94  
   640x480       75.00    72.81    59.94  
DP-1-0 disconnected (normal left inverted right x axis y axis)
DP-1-1 disconnected (normal left inverted right x axis y axis)
DP-1-2 disconnected (normal left inverted right x axis y axis)
```

However, ultimately it would be nice to expand the 'f' font size toggle button to cycle through the various resolutions. This would really require xrandr to be installed. Or maybe there is some other code with a similiar function to xrandr but without the xorg dependency.