PartialVolume/shredos.x86_64 #330: Issue After booting shredOS, error while waiting to initialize all usb devices?

U2tooo, 2025-02-12:
today when i boot ShredOs after it finish loading and enter the section where it says waiting to initialize all usb devices i get this error, after the error pops up shredos keeps in an infinite restart loop and if have to turn off the PC.
Any help would be appreciated.

images of the error below:
[https://imgur.com/a/vxb0ayd](url)

PartialVolume, 2025-02-13:
Your images show the message `Unable to create qla2xxx trace instance which I indicates you are using an earlier release. Can you try with the latest release, thanks.

^[[3~ constantly repeating, I could well be wrong but my first thought is a stuck key.

U2tooo, 2025-02-21:
hi, and sorry for the late reply, i follow your advice and do a few tests:

1st. i check all the keys in the pc and take them out one by one, and put it back, i try testing using the keyboard for normal use and no key stroke seems to be stuck, i was able to write normally, the issue still happening regardless.... 

2nd.i try with different versions of ShredOS ,  v0.38 ,  v0.37 ,  v0.36 each one with a different USB, with 0.36 get the same result at first and a few times, the with others got different errors screens, (post images below and some video)

3rd. i notice that if i disable secure boot in the laptop this issue happens more often, especially with 0.36 using secure but throw me other issues, if i use legacy boot it stops but still get some errors, like the images bellow.

with UEFI boot this issue happens, but with later tests and versions, I get a scrolling text see video bellow.

sometimes it works but most of the time I'm getting these errors, i not know what happens it was working like charm before but suddenly stops which let me to the next question:

before this happens i was cleaning a HDisk that could possibly have malware using the first USB, which raise my question can malware infect shredOS while you are using it to secure erase the HDsik, and possible the USB get infected, and the malware is causing these issues... 

but the thing is this errors are still happening in other versions 0.38, 0.37, 0.36 so far i gotten a different type of error, but i cannot reproduce some of the all the time.

at the same time trying to discard any possible issue with the laptop.

images bellow the screen boot errors i get with the tests

https://imgur.com/a/g0vwN9I

video link bellow

https://vimeo.com/1058881766/c76c8eb22c?share=copy

PartialVolume, 2025-02-21:
Your hardware definitely seems to be faulty, specifically your delete key is sending the code ^[[3^[[3 continuously. You need to fix or replace either the keyboard or maybe the keyboard I/O chip has a fault on the motherboard, this causes nwipe to start up and exit immediately and the endless scrolling after nwipe exits. I suppose it could be some weird fault with some keyboard driver, but I've never seen any faults with a keyboard driver like that.

>1st. i check all the keys in the pc and take them out one by one, and put it back, i try testing using the keyboard for normal use and no key stroke seems to be stuck, i was able to write normally, the issue still happening regardless....

When you say "I was able to write normally" how did you exactly test this? Did you boot another O.S?

>before this happens i was cleaning a HDisk that could possibly have malware using the first USB, which raise my question can malware infect shredOS while you are using it to secure erase the HDsik, and possible the USB get infected, and the malware is causing these issues...

Unless you booted the disk with malware on, the malware can't get from a disk without some autorun mechanism, something Linux doesn't do unlike Windows. The vast majority of malware will only run on Windows, so not very likely. 

I can simulate the same fault if I hold down the delete key on my laptop.

https://github.com/user-attachments/assets/86ecdd8d-8a9f-4c84-94f8-574ddf24fa97

PartialVolume, 2025-02-21:
Also booting a normal normal distribution is not a good test of whether your keyboard delete key is sending out a continuous stream of whitespace characters.

Case in point, if I boot KDE Neon and once it has booted into KDE if I hold the del key down and type normally everything works as expected even though I'm holding down the delete key while simultaneously typing. Even login works normally.

However if you switch to a virtual terminal in your normal linux distro then type the command showkey and it prints out a repeating code such as 111 then your delete key is definitely stuck even though KDE is behaving normally.

As for Windows, I've no idea how that behaves when there is a stuck delete key.

PartialVolume, 2026-02-14:
Closing as no recent activity.

closed 2026-02-14 by PartialVolume