PartialVolume/shredos.x86_64 #332: nwipe getting out of memory error

rdellaripa, 2025-02-19:
So I boot fine, set parameters and get nwipe running, and then, after a while, it gets killed with an out of memory error. I don't think it's the drive, I managed to do a dd through the whole thing with no errors.

```
[21103.027907] ip invoked oom-killer: gfp_mask=0xcc0(GFP_KERNEL), order=2, oom_score_adj=0
[21103.027929] CPU: 0 UID: 0 PID: 28563 Comm: ip Tainted: G        W        N 6.11.11 #2
[21103.027938] Tainted: [W]=WARN, [N]=TEST
[21103.027940] Hardware name: Hewlett-Packard HP Pavilion 17 Notebook PC /226A, BIOS F.43 08/19/2015
[21103.027944] Call Trace:
[21103.027949]  <TASK>
[21103.027953]  dump_stack_lvl+0x53/0x70
[21103.027965]  dump_header+0x3f/0x1a0
[21103.027973]  oom_kill_process+0xfd/0x210
[21103.027979]  out_of_memory+0x242/0x570
[21103.027985]  __alloc_pages_slowpath.constprop.0+0xac3/0xce0
[21103.027995]  __alloc_pages_noprof+0x2a7/0x310
[21103.028001]  rtl_open+0x1ae/0x4c0
[21103.028009]  __dev_open+0xe4/0x180
[21103.028015]  __dev_change_flags+0x1ac/0x220
[21103.028020]  dev_change_flags+0x21/0x60
[21103.028026]  devinet_ioctl+0x30e/0x7c0
[21103.028033]  inet_ioctl+0x1c5/0x1e0
[21103.028041]  sock_do_ioctl+0x77/0x140
[21103.028048]  sock_ioctl+0x208/0x330
[21103.028053]  __x64_sys_ioctl+0x92/0xd0
[21103.028059]  do_syscall_64+0x54/0x110
[21103.028064]  entry_SYSCALL_64_after_hwframe+0x76/0x7e
[21103.028073] RIP: 0033:0x7f85a2576938
[21103.028080] Code: 00 00 48 8d 44 24 08 48 89 54 24 e0 48 89 44 24 c0 48 8d 44 24 d0 48 89 44 24 c8 b8 10 00 00 00 c7 44 24 b8 10 00 00 00 0f 05 <89> c2 3d 00 f0 ff ff 77 07 89 d0 c3 0f 1f 40 00 48 8b 15 b1 54 0d
[21103.028084] RSP: 002b:00007fff08faf588 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[21103.028091] RAX: ffffffffffffffda RBX: 00007f85a26f6eea RCX: 00007f85a2576938
[21103.028095] RDX: 00007fff08faf600 RSI: 0000000000008914 RDI: 0000000000000003
[21103.028099] RBP: 0000000000000001 R08: 000000000000000b R09: 0000000000000803
[21103.028102] R10: 0000000000000012 R11: 0000000000000246 R12: 0000000000000003
[21103.028105] R13: 00007fff08faf600 R14: ffffffffffffffff R15: 00007fff08faf7a0
[21103.028109]  </TASK>
[21103.028112] Mem-Info:
[21103.028116] active_anon:29456 inactive_anon:179120 isolated_anon:0
                active_file:0 inactive_file:1432291 isolated_file:0
                unevictable:0 dirty:26380 writeback:0
                slab_reclaimable:71899 slab_unreclaimable:6403
                mapped:1850 shmem:205597 pagetables:299
                sec_pagetables:0 bounce:0
                kernel_misc_reclaimable:0
                free:24153 free_pcp:1877 free_cma:0
[21103.028126] Node 0 active_anon:117824kB inactive_anon:716480kB active_file:0kB inactive_file:5729164kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:7400kB dirty:105520kB writeback:0kB shmem:822388kB writeback_tmp:0kB kernel_stack:2092kB pagetables:1196kB sec_pagetables:0kB all_unreclaimable? no
[21103.028135] Node 0 DMA free:14744kB boost:0kB min:20kB low:32kB high:44kB reserved_highatomic:0KB active_anon:148kB inactive_anon:468kB active_file:0kB inactive_file:0kB unevictable:0kB writepending:0kB present:15996kB managed:15360kB mlocked:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB
[21103.028146] lowmem_reserve[]: 0 1919 6581 0
[21103.028154] Node 0 DMA32 free:55628kB boost:0kB min:3024kB low:4988kB high:6952kB reserved_highatomic:0KB active_anon:124kB inactive_anon:0kB active_file:0kB inactive_file:1781680kB unevictable:0kB writepending:28020kB present:2031428kB managed:1965692kB mlocked:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB
[21103.028164] lowmem_reserve[]: 0 0 4662 0
[21103.028172] Node 0 Normal free:26240kB boost:0kB min:7344kB low:12116kB high:16888kB reserved_highatomic:0KB active_anon:117552kB inactive_anon:716076kB active_file:0kB inactive_file:3948460kB unevictable:0kB writepending:77188kB present:5226496kB managed:5029416kB mlocked:0kB bounce:0kB free_pcp:7484kB local_pcp:0kB free_cma:0kB
[21103.028187] lowmem_reserve[]: 0 0 0 0
[21103.028193] Node 0 DMA: 78*4kB (M) 56*8kB (M) 38*16kB (M) 16*32kB (M) 3*64kB (M) 1*128kB (M) 1*256kB (M) 0*512kB 2*1024kB (UM) 1*2048kB (U) 2*4096kB (M) = 14744kB
[21103.028221] Node 0 DMA32: 11571*4kB (UE) 1185*8kB (UE) 0*16kB 0*32kB 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 55764kB
[21103.028241] Node 0 Normal: 6023*4kB (UM) 336*8kB (UM) 0*16kB 0*32kB 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 26780kB
[21103.028260] 1637931 total pagecache pages
[21103.028262] 0 pages in swap cache
[21103.028264] Free swap  = 0kB
[21103.028266] Total swap = 0kB
[21103.028267] 1818480 pages RAM
[21103.028269] 0 pages HighMem/MovableOnly
[21103.028270] 65863 pages reserved
[21103.028271] 0 pages cma reserved
[21103.028273] Tasks state (memory values in pages):
[21103.028275] [  pid  ]   uid  tgid total_vm      rss rss_anon rss_file rss_shmem pgtables_bytes swapents oom_score_adj name
[21103.028280] [    145]     0   145     1049      668       32        0       636    49152        0             0 syslogd
[21103.028288] [    149]     0   149     1049      635       32        0       603    49152        0             0 klogd
[21103.028294] [    164]     0   164      629      231        0        0       231    45056        0             0 thd
[21103.028300] [    171]   100   171     1175      301       33        0       268    53248        0             0 dbus-daemon
[21103.028306] [    175]     0   175      670      330       32        0       298    45056        0             0 rpcbind
[21103.028312] [    180]     0   180     1091      777       64        0       713    49152        0             0 shredos_net.sh
[21103.028318] [    186]   101   186    19262      756      205        0       551    57344        0             0 ntpd
[21103.028324] [    192]     0   192     1049      619       32        0       587    45056        0             0 crond
[21103.028329] [    200]     0   200      689      101        0        0       101    45056        0             0 tftpd
[21103.028335] [    206]     0   206      945      352       46        0       306    45056        0             0 collectd
[21103.028340] [    207]     0   207     1049      642       32        0       610    49152        0             0 getty
[21103.028346] [    208]     0   208     1124      786       96        0       690    53248        0             0 nwipe_launcher
[21103.028352] [    209]     0   209     1157      902       96        0       806    53248        0             0 sh
[21103.028358] [    458]     0   458    43839     2618     1824        0       794    81920        0             0 nwipe
[21103.028364] [  28560]     0 28560     1049      623       32        0       591    49152        0             0 ifup
[21103.028369] [  28563]     0 28563     1049      564       32        0       532    45056        0             0 ip
[21103.028375] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=nwipe,pid=458,uid=0
[21103.028414] Out of memory: Killed process 458 (nwipe) total-vm:175356kB, anon-rss:7296kB, file-rss:0kB, shmem-rss:3176kB, UID:0 pgtables:80kB oom_score_adj:0
```

PartialVolume, 2025-02-19:
IP invoked the oom-killer

Nwipe was killed just because it will be the biggest consumer of memory even though the out of memory might not be anything to do with nwipe.

As IP invoked the oom-killer maybe it's something to do with the network. Can you plug an ethernet lead in and connect it to a network running a DHCP server. I'm wondering whether IP software running in the background is causing a memory leak. If plugging into a network fixes the problem then it will give me an idea where to look.

PartialVolume, 2025-02-19:
Something else you can do to determine whether it's a network issue.

Once nwipe has appeared, type ALT F2 to switch virtual terminals and run pkill shredos_net.sh, then ALT F1 and start the wipes. See if that then works without the out of memory. This will kill the software that manages the ethernet.

rdellaripa, 2025-02-19:
I have it running right now with a network connection. If that fails, I'll try the rest.

rdellaripa, 2025-02-24:
Well, it worked unplugged. Now I can't get it to fail again. I'll update again when I find some consistency.

PartialVolume, 2025-02-26:
> I'm afraid I don't have any technical details, but it seems to correlate to choosing Lagged Fibonacci as the PRNG.

I'm going to ask @Knogle to comment as the author of the lagged Fibonacci PRNG.

For what it's worth I've been using lagged  Fibonacci to wipe at least 50 drives on two different systems, Dell I7 8 core 8GB (8 drives simultaneously) and a Clevo I7 20 core (one drive at a time) and haven't experienced any out of memory errors.

@rdellaripa  Can you post the output of `lspci -k` so we can maybe find a correlation with the hardware. Thanks.

rdellaripa, 2025-02-26:
00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 16h (Models 30h-3fh) Processor Root Complex
	Subsystem: Hewlett-Packard Company Device 226a
00:01.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Mullins [Radeon R4/R5 Graphics] (rev 05)
	DeviceName: Onboard IGD
	Subsystem: Hewlett-Packard Company Device 226a
	Kernel driver in use: radeon
00:01.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Kabini HDMI/DP Audio
	Subsystem: Hewlett-Packard Company Device 226a
00:02.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 16h (Models 30h-3fh) Host Bridge
00:02.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Family 16h Processor Functions 5:1
	Subsystem: Hewlett-Packard Company Device 226a
	Kernel driver in use: pcieport
00:02.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Family 16h Processor Functions 5:1
	Subsystem: Hewlett-Packard Company Device 226a
	Kernel driver in use: pcieport
00:02.3 PCI bridge: Advanced Micro Devices, Inc. [AMD] Family 16h Processor Functions 5:1
	Subsystem: Hewlett-Packard Company Device 226a
	Kernel driver in use: pcieport
00:02.5 PCI bridge: Advanced Micro Devices, Inc. [AMD] Family 16h Processor Functions 5:1
	Subsystem: Hewlett-Packard Company Device 226a
	Kernel driver in use: pcieport
00:08.0 Encryption controller: Advanced Micro Devices, Inc. [AMD] Kabini/Mullins PSP-Platform Security Processor
	Subsystem: Hewlett-Packard Company Device 226a
00:10.0 USB controller: Advanced Micro Devices, Inc. [AMD] FCH USB XHCI Controller (rev 11)
	Subsystem: Hewlett-Packard Company Device 226a
	Kernel driver in use: xhci_hcd
00:11.0 SATA controller: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] (rev 40)
	Subsystem: Hewlett-Packard Company Device 226a
	Kernel driver in use: ahci
00:12.0 USB controller: Advanced Micro Devices, Inc. [AMD] FCH USB EHCI Controller (rev 39)
	Subsystem: Hewlett-Packard Company Device 226a
	Kernel driver in use: ehci-pci
00:13.0 USB controller: Advanced Micro Devices, Inc. [AMD] FCH USB EHCI Controller (rev 39)
	Subsystem: Hewlett-Packard Company Device 226a
	Kernel driver in use: ehci-pci
00:14.0 SMBus: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller (rev 42)
	Subsystem: Hewlett-Packard Company Device 226a
00:14.2 Audio device: Advanced Micro Devices, Inc. [AMD] FCH Azalia Controller (rev 02)
	Subsystem: Hewlett-Packard Company Device 226a
00:14.3 ISA bridge: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge (rev 11)
	Subsystem: Hewlett-Packard Company Device 226a
00:18.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 16h (Models 30h-3fh) Processor Function 0
00:18.1 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 16h (Models 30h-3fh) Processor Function 1
00:18.2 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 16h (Models 30h-3fh) Processor Function 2
00:18.3 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 16h (Models 30h-3fh) Processor Function 3
00:18.4 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 16h (Models 30h-3fh) Processor Function 4
00:18.5 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 16h (Models 30h-3fh) Processor Function 5
02:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8188EE Wireless Network Adapter (rev 01)
	DeviceName: Realtek Focus RTL8188EE bgn 1x1 PCI-e HMC WW
	Subsystem: Hewlett-Packard Company RTL8188EE mini-PCIe card
03:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL810xE PCI Express Fast Ethernet controller (rev 07)
	DeviceName: Realtek PCIe FE Family Controller
	Subsystem: Hewlett-Packard Company Device 226a
	Kernel driver in use: r8169
04:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS5229 PCI Express Card Reader (rev 01)
	DeviceName: Realtek PCIE CardReader
	Subsystem: Hewlett-Packard Company Device 226a

Knogle, 2025-02-28:
I hope you are doing fine :) Unfortunately I think it has nothing to do with nwipe. According to the trace, the OOM Killer was invoked by "ip" responsible for networking.
The nwipe process has 175M in use at this time. So I don't expect any issue from there.

rdellaripa, 2025-03-02:
Continuing to test. Just throwing these out here in case they're helpful later.

/var/log/messages:
[13229.407609] ip invoked oom-killer: gfp_mask=0xcc0(GFP_KERNEL), order=2, oom_score_adj=0
[13229.407631] CPU: 1 UID: 0 PID: 3061 Comm: ip Tainted: G        W        N 6.11.11 #2
[13229.407639] Tainted: [W]=WARN, [N]=TEST
[13229.407641] Hardware name: Hewlett-Packard HP Pavilion 17 Notebook PC /226A, BIOS F.43 08/19/2015
[13229.407645] Call Trace:
[13229.407651]  <TASK>
[13229.407655]  dump_stack_lvl+0x53/0x70
[13229.407666]  dump_header+0x3f/0x1a0
[13229.407674]  oom_kill_process+0xfd/0x210
[13229.407680]  out_of_memory+0x242/0x570
[13229.407686]  __alloc_pages_slowpath.constprop.0+0xac3/0xce0
[13229.407694]  __alloc_pages_noprof+0x2a7/0x310
[13229.407700]  rtl_open+0x1ae/0x4c0
[13229.407709]  __dev_open+0xe4/0x180
[13229.407715]  __dev_change_flags+0x1ac/0x220
[13229.407720]  dev_change_flags+0x21/0x60
[13229.407725]  devinet_ioctl+0x30e/0x7c0
[13229.407733]  inet_ioctl+0x1c5/0x1e0
[13229.407740]  sock_do_ioctl+0x77/0x140
[13229.407747]  sock_ioctl+0x208/0x330
[13229.407752]  __x64_sys_ioctl+0x92/0xd0
[13229.407758]  do_syscall_64+0x54/0x110
[13229.407763]  entry_SYSCALL_64_after_hwframe+0x76/0x7e
[13229.407772] RIP: 0033:0x7f1f2a5a9938
[13229.407778] Code: 00 00 48 8d 44 24 08 48 89 54 24 e0 48 89 44 24 c0 48 8d 44 24 d0 48 89 44 24 c8 b8 10 00 00 00 c7 44 24 b8 10 00 00 00 0f 05 <89> c2 3d 00 f0 ff ff 77 07 89 d0 c3 0f 1f 40 00 48 8b 15 b1 54 0d
[13229.407783] RSP: 002b:00007ffc7f2ab048 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[13229.407790] RAX: ffffffffffffffda RBX: 00007f1f2a729eea RCX: 00007f1f2a5a9938
[13229.407793] RDX: 00007ffc7f2ab0c0 RSI: 0000000000008914 RDI: 0000000000000003
[13229.407797] RBP: 0000000000000001 R08: 000000000000000b R09: 0000000000000803
[13229.407800] R10: 0000000000000012 R11: 0000000000000246 R12: 0000000000000003
[13229.407803] R13: 00007ffc7f2ab0c0 R14: ffffffffffffffff R15: 00007ffc7f2ab260
[13229.407807]  </TASK>
[13229.407810] Mem-Info:
[13229.407813] active_anon:29493 inactive_anon:178787 isolated_anon:0
                active_file:16 inactive_file:1431402 isolated_file:0
                unevictable:0 dirty:50114 writeback:0
                slab_reclaimable:72284 slab_unreclaimable:6445
                mapped:2481 shmem:205539 pagetables:337
                sec_pagetables:0 bounce:0
                kernel_misc_reclaimable:0
                free:25086 free_pcp:608 free_cma:0
[13229.407824] Node 0 active_anon:117972kB inactive_anon:715148kB active_file:64kB inactive_file:5725608kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:9924kB dirty:200456kB writeback:0kB shmem:822156kB writeback_tmp:0kB kernel_stack:2144kB pagetables:1348kB sec_pagetables:0kB all_unreclaimable? no
[13229.407833] Node 0 DMA free:13952kB boost:0kB min:20kB low:32kB high:44kB reserved_highatomic:0KB active_anon:264kB inactive_anon:1132kB active_file:0kB inactive_file:8kB unevictable:0kB writepending:8kB present:15996kB managed:15360kB mlocked:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB
[13229.407844] lowmem_reserve[]: 0 1919 6581 0
[13229.407852] Node 0 DMA32 free:63784kB boost:0kB min:3024kB low:4988kB high:6952kB reserved_highatomic:0KB active_anon:140kB inactive_anon:692kB active_file:0kB inactive_file:1771508kB unevictable:0kB writepending:193188kB present:2031428kB managed:1965692kB mlocked:0kB bounce:0kB free_pcp:384kB local_pcp:0kB free_cma:0kB
[13229.407862] lowmem_reserve[]: 0 0 4662 0
[13229.407869] Node 0 Normal free:22608kB boost:0kB min:7344kB low:12116kB high:16888kB reserved_highatomic:0KB active_anon:117576kB inactive_anon:713532kB active_file:64kB inactive_file:3953628kB unevictable:0kB writepending:6512kB present:5226496kB managed:5029416kB mlocked:0kB bounce:0kB free_pcp:2024kB local_pcp:0kB free_cma:0kB
[13229.407879] lowmem_reserve[]: 0 0 0 0
[13229.407886] Node 0 DMA: 111*4kB (M) 61*8kB (M) 32*16kB (M) 13*32kB (M) 1*64kB (M) 2*128kB (M) 0*256kB 1*512kB (M) 1*1024kB (U) 1*2048kB (U) 2*4096kB (M) = 13956kB
[13229.407917] Node 0 DMA32: 13559*4kB (UE) 1222*8kB (UE) 0*16kB 0*32kB 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 64012kB
[13229.407938] Node 0 Normal: 3554*4kB (UM) 1130*8kB (UM) 0*16kB 0*32kB 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 23256kB
[13229.407960] 1636992 total pagecache pages
[13229.407962] 0 pages in swap cache
[13229.407963] Free swap  = 0kB
[13229.407965] Total swap = 0kB
[13229.407967] 1818480 pages RAM
[13229.407968] 0 pages HighMem/MovableOnly
[13229.407970] 65863 pages reserved
[13229.407972] 0 pages cma reserved
[13229.407973] Tasks state (memory values in pages):
[13229.407975] [  pid  ]   uid  tgid total_vm      rss rss_anon rss_file rss_shmem pgtables_bytes swapents oom_score_adj name
[13229.407981] [    143]     0   143     1049      616       32        0       584    49152        0             0 syslogd
[13229.407989] [    147]     0   147     1049      618       32        0       586    45056        0             0 klogd
[13229.407996] [    163]     0   163      629      231        0        0       231    40960        0             0 thd
[13229.408002] [    170]   100   170     1175      329       33        0       296    49152        0             0 dbus-daemon
[13229.408009] [    174]     0   174      670      329       32        0       297    49152        0             0 rpcbind
[... cut ...]
SUnreclaim:        25792 kB
KernelStack:        2140 kB
PageTables:         1484 kB
SecPageTables:         0 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:     3505232 kB
Committed_AS:     838524 kB
VmallocTotal:   34359738367 kB
VmallocUsed:       12244 kB
VmallocChunk:          0 kB
Percpu:             1440 kB
CmaTotal:              0 kB
CmaFree:               0 kB
DirectMap4k:       16196 kB
DirectMap2M:     5160960 kB
DirectMap1G:     2097152 kB

rdellaripa, 2025-03-06:
Further testing has shown that this issue occurs with ISAAC and ISAAC64 as well as lagged Fibonacci. I have not been able to reproduce it on Mersene Twister or XO Rohsiro.

DerGregor, 2025-03-07:
I would also like to join in here. I have exactly the same "ip invoked oom-killer" problem.
see attached log.
the only change in the settings was the wipe method to "Fill with Zeros".

> As IP invoked the oom-killer maybe it's something to do with the network. Can you plug an ethernet lead in and connect it to a network running a DHCP server. I'm wondering whether IP software [...]

I have disabled the onboard LAN adapter in the BIOS. 
so far the wipe process is running. I have never gotten above 0.3% before. i am now at 1.5%. 

[dmesg_2025-03-05-11-52-53_system_uuid_00000000-0000-0000-0000-4ccc6ad83f18.txt](https://github.com/user-attachments/files/19123773/dmesg_2025-03-05-11-52-53_system_uuid_00000000-0000-0000-0000-4ccc6ad83f18.txt)

Knogle, 2025-03-07:
Can you maybe run `ip -V` please, i'll try to find out if there is some sort of issue with ip.

DerGregor, 2025-03-07:
> Can you maybe run `ip -V` please, i'll try to find out if there is some sort of issue with ip.

i hope i understand you correctly. i opened another console with ALT + F2 and entered ip -V. but this parameter does not seem to exist.

[image]

DerGregor, 2025-03-07:
short update. system is still running and stable with deactivated network adapter.
[image]
I will let the system run over the weekend and report back if it was successful. I may then run it again with the network adapter activated and an active network connection.

Is there a reason why ShredOS comes with network support?

PartialVolume, 2025-03-08:
Whether you use the network is entirely optional however ShredOS supports NTP (Network Time Protocol) so if you are plugged into a network with internet access it will attempt to contact a NTP server to check what the date and time is and adjust it if the system clock is wrong. If you don't want to connect to a network then you can adjust the system date and time via Nwipe's config menu 'c' (accessed via the drive selection screen) or alternatively via the command line.

Network access is required if you are PXE booting.

Network accessed is also required if you want to upload your PDFs, logs automatically to a lftp, FTP or sftp server.

In addition you can store Nwipe's config and customer files remotely so if you are running a hundred ShredOS systems they all bootup with the same customer list and config. This is configured via the shredos_output=".." and shredos_config=".." commands that are placed on the kernel command line in grub.cfg

PartialVolume, 2025-03-08:
>maybe it really only happens when you are offline and not with an active connection

That's a possibility, I generally run ShredOS with an active internet enabled network on my desktop systems however I run laptop wipes without a network connection. When I get a chance I'll run it without a network connection and see if I can reproduce the out of memory. It might also be specific to your hardware as well as only happening with an active network connection. I need to take a closer look at your lspci -k output.

PartialVolume, 2025-03-08:
One other thought, when there is no active network connection the program called 'ip' is executed periodically, I don't remember how often whether every 5 seconds or 30 seconds to check the ethernet port status. This polling doesn't occur once a IP address has been requested and received from the DHCP server so it's possible that if the program called 'ip' is not freeing up allocated memory on exit then slowly memory will get used up.

As noted from previous comments about the -V option, the version of 'ip' being used is the stripped down busybox version. Just something to bear in mind when searching for the root cause of this issue.

DerGregor, 2025-03-08:
> I need to take a closer look at your lspci -k output.

yeh, I'll provide it on Monday (with enabled LAN adapter)

> This polling doesn't occur once a IP address has been requested and received from the DHCP server so it's possible that if the program called 'ip' is not freeing up allocated memory on exit then slowly memory will get used up.

I could also test that on monday. I still have a few old routers lying around here so at least there would be a network to connect to, albeit without internet.

rdellaripa, 2025-03-08:
I'm actually running a wipe right now connected just to an old switch so it doesn't have DCHP, internet, etc.

What I don't understand at this point is that I've run top -oVIRT in another console, and I haven't seen ip just hanging out there slowly eating up memory.  But it is obvious something is, because MemFree goes from 50704 kB to 6101760 kB free when the oom triggers...despite nwipe (or ntpd, the second largest process) growing in size over time. I thought maybe a device driver wasn't freeing memory, but lsmod doesn't show anything.

DerGregor, 2025-03-10:
so first of all good news. switching off the NIC helped. my disks were successfully zeroed. Duration: approx. 44-45 hours.
today I reactivated the NIC in the BIOS and started a wipe process again. as expected, the process failed again after 5-10 minutes.
here is another log: 
[dmesg_2025-03-10-09-14-23_system_uuid_00000000-0000-0000-0000-4ccc6ad83f18.txt](https://github.com/user-attachments/files/19159430/dmesg_2025-03-10-09-14-23_system_uuid_00000000-0000-0000-0000-4ccc6ad83f18.txt)
here is the requested lspci -k output

[lspci.txt](https://github.com/user-attachments/files/19159455/lspci.txt)

> This polling doesn't occur once a IP address has been requested and received from the DHCP server so it's possible that if the program called 'ip' is not freeing up allocated memory on exit then slowly memory will get used up.

I have now started a new wipe process but this time I have connected an old TP-Link router. a network with DHCP etc. but without internet.
The process has been running smoothly so far. at least for 30 minutes. that's a good sign for me. because the system never ran that long when the process crashed.

rdellaripa, 2025-03-10:
Attached is the output of
```
cat /proc/meminfo
slabtop -o -s c | head -20
tail -25 /var/log/messages
```
run once a minute, snipped to the minute before and after the oom.
again, most of the memory seems to be in buffers. I'm not sure at the moment how to dig deeper on that.

[meminfo-2025-03-10.txt](https://github.com/user-attachments/files/19167807/meminfo-2025-03-10.txt)

PartialVolume, 2025-03-10:
Can you try killing the ntpd process and dhcpd process in turn in order to try and identify which one is causing the issue.

With no network connection I still can't reproduce the issue on my laptop. I was reading about some buffer overflow problem in ntpd but that was way back in the early 2000s. Triggered by a rogue ntp server. So maybe worth killing those processes to see if they are the cause.

rdellaripa, 2025-03-11:
I killed the ntpd process and the issue recurred. I can't kill the dhcpd process, as its spawned from an "if up" from shredos_net.sh

rdellaripa, 2025-03-12:
The one I have running now with shredos_net.sh killed has been running over a day without failure.

rdellaripa, 2025-03-13:
Ran successfully, completing in 40 hours. Having two workarounds (plugging the NIC into something or killing shredos_net.sh), I don't need anything more. You can close it, or leave it open until you do something like have the net.sh have a timeout, up to you.

DerGregor, 2025-03-13:
Good to hear that it worked. However, I don't think this issue should be closed. there is obviously a bug here, and personally I don't want to turn off the NIC in the BIOS, or stick something in the NIC, or kill a process when I want to start a wipe process.
>  something like have the net.sh have a timeout
a timeout would be a suitable solution. Or maybe you could ask the user at startup if they want network support or not.

PartialVolume, 2025-03-13:
Yes, If you could leave this open.

With no ethernet connection and NIC enabled in the BIOS can you boot ShredOS and tell me what the output is if you run the following commands

```
cd /sys/class/net

ls ( should output your network interfaces, cd to the ethernet one
enp0s31f6  lo  wlp0s20f3 these are mine.

cd enp0s31f6  (or whatever yours is called)

ls (should show the following)
addr_assign_type    device             ifindex               phys_port_id    subsystem
address             dev_id             iflink                phys_port_name  testing
addr_len            dev_port           link_mode             phys_switch_id  threaded
broadcast           dormant            mtu                   power           tx_queue_len
carrier             duplex             name_assign_type      proto_down      type
carrier_changes     flags              napi_defer_hard_irqs  queues          uevent
carrier_down_count  gro_flush_timeout  netdev_group          speed
carrier_up_count    ifalias            operstate             statistics

then type:

more carrier
0 (should be 0 with no network lead plugged in)

more operstate
down (should say down with no network lead plugged)

Let me know what the output of those two more commands are. Thanks
```

PartialVolume, 2025-03-13:
Something I do need to change is how it fills up the log with network down messages. I'll change the code so that it repeats the message five times max then goes quiet until a state change, (a network cable being plugged in). I don't think this has anything to do with the out of memory but will just make viewing the log a lot easier.

DerGregor, 2025-03-14:
> more carrier
0 (should be 0 with no network lead plugged in)

> more operstate
down (should say down with no network lead plugged)

I checked it and both values are as you wrote

rdellaripa, 2025-03-14:
carrier is 0 and operstate is down.

kduvekot, 2025-04-20:
I am running ShredOS v2024.11_27_x86-64_0.38 (but probably 0.38 should be enough I think) .. and I have the OOM problem on 2 seperate laptops. 

One is a ASUS with 8Gb memory, the other an Sony VAIO with 6Gb memory.

Let me know what you would like to me do/export/produce etc .. 
on the Asus I  think I can disable the network adapter (and WiFi) from the BIOS .. on the VAIO I can't.

I am going to run some tests .. and I have these laptops around for a while if they can help with testing let me know

PartialVolume, 2025-04-20:
@kduvekot Can you confirm when the out of memory occurs in the boot process. Is it immediately after the kernel has loaded to 100% and before any other text is displayed?

kduvekot, 2025-04-21:
@PartialVolume it happens during the wipe, usually after a while .. so like an hour into the wipe.

On the ASUS it didn't happen yet when I disabled the network in the BIOS ( 2 x 750Gb discs .. taking 50+ hours with DoD method) .. it still going 

On the VAIO I killed the shredos_net.sh proces and it is also still going on .. (20+ hour wipe)

(Those were the two suggested workarounds in this thread)

If you like I can not do the workarounds and run some monitoring script to see what happens over time...

DerGregor, 2025-04-21:
the problem does not occur with every system. 
@kduvekot which NIC do you have on the VAIO and ASUS device? for me and rdellaripa it's a NIC from Realtek.

DerGregor, 2025-04-21:
in that case i would assume that your ASUS device also uses some kind of realtek chip. presumably this error only occurs on devices where linux loads the (r8169) realtek driver?

kduvekot, 2025-04-21:
so that sounds like something that can be solvable :-)  ... let wait and see after the VAIO is finished  what I can do to replicate it .. and what data I can collect. Maybe just like described earliers .. run a small script that gets some memory statistics every minut or something like that ..

PartialVolume, 2025-04-21:
Here's one test you could do or maybe have already done:

So killing shredos_net.sh fixes the out of memory. However I'm assuming this without being plugged into a network providing DHCP.

So my question is if you do plug the laptop into a network providing DHCP then it works ok without any out of memory? I know this was indeed the case on earlier posts. I just want confirmation you are seeing the same behaviour.

While it could quite possibly be a memory leak in the r8169 driver that needs fixing it does appear my network script is triggering this fault. With that in mind I'll look at modifying the network script to see if I can solve the problem there first. If there is no network cable plugged in, the script does poll that the status and run DHCP request every few seconds, something that doesn't seem to be a problem with most NICS.

If you are up for testing a new built I can see if there are some modifications to the script I can make. No promises on when I can do this as I'm in the middle of a house move and up to my eyeballs in fixing the plumbing at the moment! 😊

kduvekot, 2025-04-21:
I have restarted the VAOI laptop .. and just let it sit on the starting screen if nwipe ..
switched to ALT-F2 console and started top.

after 5 hour and 45 min .. nothing has broken yet .. but .. the memory usage of nwipe has risen from 94m to 145m. the second largest memory usage is by ntp (75m) but stable all the time.

shredos_net.sh usage is roughly 4.3m with some additional memory used by the child processes but nothing out of the ordinary 

any sugestions to do now? .. I was thinking about starting a single run of a zero wipe in nwipe and see what happens with the memory of nwipe during that run.

kduvekot, 2025-04-22:
the ASUS laptop has finished wiping .. and it contains a Qualcomm Atheros AR8161 Gigabit Ethernet adapter (and a AR9485 WiFI adapter) .. 
I have enabled them again .. and started a DoD wipe again .. to see when it fails this time .. and monitoring top at the same time (I hope to catch the crash)

the ASUS does not contain the Ethernet adapter messages in the dmesg output that are present on the VAIO with the Realtek adapter .. but from a process view it looks like its doing the exact same thing (and same memory usage for the other processes than nwipe)

kduvekot, 2025-04-22:
FYI .. I did not connect any network cables to these laptops yet .. just trying to replicate the issue that I had before.

kduvekot, 2025-04-22:
So both laptops crashed with OOM .. so good to know I can still replicate the issue.

on the ASUS it crashed after 5077 seconds of uptime (based on dmesg timestamp) doing a DoD wipe
on the VAOI it crashed after 82655 seconds o uptime (almost 23 hours, but you can subtract the 21 hours of doing nothing from it)

so now the question is .. what script should I run to keep track what is happening .. and make sure that it is stored on the disc :-)  .. let me know what helps the most.

I will try the ASUS with the network cable connected to see if that prevents the crash from happening now ..

kduvekot, 2025-04-23:
the ASUS laptop has been running for over 16 hours without any issues with the network cable connected to a network that provides DHCP. so that workaround has been confirmed as far as I tell right now,

kduvekot, 2025-05-10:
@PartialVolume how is the plumbing going? What else would you like me to test? And any ideas what has caused this issue? Looks like it happened between .37 and .38 .. and not related to nwipe itself it looks like. So something in the buildroot setup I think.

PartialVolume, 2025-05-13:
> [@PartialVolume](https://github.com/PartialVolume) how is the plumbing going? What else would you like me to test? And any ideas what has caused this issue? Looks like it happened between .37 and .38 .. and not related to nwipe itself it looks like. So something in the buildroot setup I think.

Apologies for the delay in replying. It's going well, although, as is often the case once you start taking things apart you find more things that could do with fixing!

I've been running a few tests on an old Sony Vaio without the RJ45 network connected. Although it's not generated any "out of memory" errors yet, it's only been running for 2-3 hours, there are a few things I'm going to change in shredos_net.sh.

Without a network cable connected every 5 seconds the current script is running the programs ifup and ifdown. ifup, in addition is triggering DHCP requests which of course fail. This constant execution of these programs is unnecessary as I think I only need to monitor a state change on the carrier or status signals when a cable gets plugged in. Only then should it be necessary to bring the network up and request DHCP. 

If this works it will reduce to zero communications with the network card when the cable is not plugged in. This is a vast reduction in communication with the network card which in itself will stop the out of memory. This will also have some beneficial knock on effects. The /var/log/messages and messages.0 messages.1 etc won't keep growing in size due to the endless network messages. In addition nohup.log, which is another network log file will no longer keep growing in size.

Once I've made the changes to this network script and generated a test .img or .iso, I'll let you know and if you could test it on your system that generates the out of memory that would be appreciated.

kduvekot, 2025-05-13:
I have 3 old laptops waiting for a test ...
One that doesn't give an error ... And two that reliable can generate the error after a while (2-3 hours I thought).

XVII, 2025-05-20:
This appears to be related to not having an internet connection. On an 8GB machine it ran out of memory within 2 minutes or so after starting the erase. Adding a network connection and it's running smoother now.

kduvekot, 2025-06-04:
> Once I've made the changes to this network script and generated a test .img or .iso, I'll let you know and if you could test it on your system that generates the out of memory that would be appreciated.

Any progress ... these testing laptops take up space on my desk 😁

gorbiWTF, 2025-07-31:
Yup, connecting the laptop to my network seems to have "fixed" the problem. Wipe went through successfully.

<details><summary>dmesg log</summary>
<p>

```
[    0.000000] Linux version 6.11.11 (nick@nick-np5xnp6xnp7xpnp) (x86_64-buildroot-linux-gnu-gcc.br_real (Buildroot -ge80f0d4972) 13.3.0, GNU ld (GNU Binutils) 2.42) #2 SMP PREEMPT_DYNAMIC Thu Jan 23 10:42:35 GMT 2025
[    0.000000] Command line: BOOT_IMAGE=/boot/shredos console=tty3 loglevel=3 "nwipe_options=--method=prng --verify=all --autopoweroff" shredos_exclude_boot_disc=yes loadkeys=de
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000008efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000008f000-0x000000000008ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000090000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000020000000-0x00000000200fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000020100000-0x00000000b8d26fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000b8d27000-0x00000000b8d56fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b8d57000-0x00000000b8d66fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000b8d67000-0x00000000b93a9fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000b93aa000-0x00000000b9b97fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b9b98000-0x00000000b9b98fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000b9b99000-0x00000000b9bdafff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b9bdb000-0x00000000b9d50fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000b9d51000-0x00000000b9ff9fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b9ffa000-0x00000000b9ffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000e00f8000-0x00000000e00f8fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffc00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000013fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] APIC: Static calls initialized
[    0.000000] e820: update [mem 0xaa60e018-0xaa61e057] usable ==> usable
[    0.000000] e820: update [mem 0xaa600018-0xaa60d057] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000008efff] usable
[    0.000000] reserve setup_data: [mem 0x000000000008f000-0x000000000008ffff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000000090000-0x000000000009dfff] usable
[    0.000000] reserve setup_data: [mem 0x000000000009e000-0x000000000009ffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x000000001fffffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000020000000-0x00000000200fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000020100000-0x00000000aa600017] usable
[    0.000000] reserve setup_data: [mem 0x00000000aa600018-0x00000000aa60d057] usable
[    0.000000] reserve setup_data: [mem 0x00000000aa60d058-0x00000000aa60e017] usable
[    0.000000] reserve setup_data: [mem 0x00000000aa60e018-0x00000000aa61e057] usable
[    0.000000] reserve setup_data: [mem 0x00000000aa61e058-0x00000000b8d26fff] usable
[    0.000000] reserve setup_data: [mem 0x00000000b8d27000-0x00000000b8d56fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000b8d57000-0x00000000b8d66fff] ACPI data
[    0.000000] reserve setup_data: [mem 0x00000000b8d67000-0x00000000b93a9fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x00000000b93aa000-0x00000000b9b97fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000b9b98000-0x00000000b9b98fff] usable
[    0.000000] reserve setup_data: [mem 0x00000000b9b99000-0x00000000b9bdafff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000b9bdb000-0x00000000b9d50fff] usable
[    0.000000] reserve setup_data: [mem 0x00000000b9d51000-0x00000000b9ff9fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000b9ffa000-0x00000000b9ffffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000e00f8000-0x00000000e00f8fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000ffc00000-0x00000000ffffffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000100000000-0x000000013fffffff] usable
[    0.000000] efi: EFI v2.3.1 by American Megatrends
[    0.000000] efi: ACPI=0xb9331000 ACPI 2.0=0xb9331000 SMBIOS=0xb9a81e18 
[    0.000000] efi: Not removing mem34: MMIO range=[0xe00f8000-0xe00f8fff] (4KB) from e820 map
[    0.000000] efi: Not removing mem35: MMIO range=[0xfed01000-0xfed01fff] (4KB) from e820 map
[    0.000000] efi: Remove mem36: MMIO range=[0xffc00000-0xffffffff] (4MB) from e820 map
[    0.000000] e820: remove [mem 0xffc00000-0xffffffff] reserved
[... 1486 lines cut ...]
[    8.633045] usb 1-4: new high-speed USB device number 3 using xhci_hcd
[    8.779905] input: PS/2 Logitech Wheel Mouse as /devices/platform/i8042/serio4/input/input14
[... cut ...]
[    9.454643] sd 2:0:0:0: [sdb] 60062500 512-byte logical blocks: (30.8 GB/28.6 GiB)
[    9.456256] sd 2:0:0:0: [sdb] Write Protect is off
[    9.456281] sd 2:0:0:0: [sdb] Mode Sense: 43 00 00 00
[    9.456908] sd 2:0:0:0: [sdb] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[    9.475609]  sdb: sdb1
[    9.476079] sd 2:0:0:0: [sdb] Attached SCSI removable disk
[   11.997368] r8169 0000:03:00.1 eth0: Link is Up - 1Gbps/Full - flow control off
[   18.534095] platform surface_aggregator_cdev: deferred probe pending: (reason unknown)
```

</p>
</details>

mwwgithub, 2025-08-17:
I think I have encountered the same issue.  Attached are logs from a few different unsuccessful attempts to wipe a pair of disks.

[nwipe_log_20250816-235840.txt](https://github.com/user-attachments/files/21822063/nwipe_log_20250816-235840.txt)
[dmesg_2025-08-16-23-58-39_system_uuid_53457cbc-7445-11e6-a0e6-2c5267241a00.txt](https://github.com/user-attachments/files/21822059/dmesg_2025-08-16-23-58-39_system_uuid_53457cbc-7445-11e6-a0e6-2c5267241a00.txt)
[dmesg_2025-08-16-22-50-07_system_uuid_53457cbc-7445-11e6-a0e6-2c5267241a00.txt](https://github.com/user-attachments/files/21822060/dmesg_2025-08-16-22-50-07_system_uuid_53457cbc-7445-11e6-a0e6-2c5267241a00.txt)
[nwipe_log_20250816-222949.txt](https://github.com/user-attachments/files/21822061/nwipe_log_20250816-222949.txt)
[dmesg_2025-08-16-22-29-48_system_uuid_53457cbc-7445-11e6-a0e6-2c5267241a00.txt](https://github.com/user-attachments/files/21822062/dmesg_2025-08-16-22-29-48_system_uuid_53457cbc-7445-11e6-a0e6-2c5267241a00.txt)
[dmesg_2025-08-17-00-52-03_system_uuid_53457cbc-7445-11e6-a0e6-2c5267241a00.txt](https://github.com/user-attachments/files/21822064/dmesg_2025-08-17-00-52-03_system_uuid_53457cbc-7445-11e6-a0e6-2c5267241a00.txt)

mwwgithub, 2025-08-17:
> > I think I have encountered the same issue. Attached are logs from a few different unsuccessful attempts to wipe a pair of disks.
> 
> Can you confirm this is without any DHCP network attached via RJ45. Looks like it is.
> 
> And your network hardware is Generic FE-GE Realtek PHY r8169-0-100:00: attached PHY driver. Realtek R8169.

I can confirm.  No Ethernet cable connected to the port.

I went on to disable the onboard NIC and no crash since.

Knogle, 2025-11-30:
It seems to be related to the Realtek 8169 NIC.

kduvekot, 2025-11-30:
@PartialVolume because the problems only happened on version 0.38 and did not happen on 0.37 it might be good to see what has changed between those two versions in relation to network drivers and scripts.

PartialVolume, 2025-11-30:
I'm just reworking the network script to only take any action when the carrier up / carrier down count changes, that should stop the repeating messages to tmpfs.

DaanSelen, 2025-12-04:
Interesting bug this is! For now I see that disabling network or setting the time right might fix the issue. But I hope to see something that would fix it from 0. :)

Knogle, 2026-02-09:
> fyi: I tried the latest (2025.11_28_x86-64_v0.40_20260204) on one of my machines where nwipe had been getting killed (`ip invoked oom-killer`) when using 2024.11_27_x86-64_0.38_20250123. Just using defaults, it succeeded. I think your fix works.

As you have a system which reliably reproduces this issue, can you maybe test the following image?
https://nextcloud.druschke.network/index.php/s/bxJRw4FCmyXiwja

This one is nwipe's 2024 build, but has a kernel patch applied. Just need someone to test in order to make sure it works, and submit it for upstream kernel.

stringTrimmer, 2026-02-09:
@Knogle  , I wasn't able to reach your website.  Could you maybe publish a special/temporary release here on github and I could download from there?

DaanSelen, 2026-02-12:
> @Knogle , I was able to get to your google drive link, but looks like its an iso.  Sorry, I don't have ready access to a dvd drive anymore to burn it, can you/I convert it to an img for writing to a usb drive?

You can write ISO to a USB

closed 2026-02-14 by PartialVolume

stringTrimmer, 2026-02-15:
@Knogle I ran your patched version with defaults and it succeeded.  Then I ran the original v2024.11_27_x86-64_0.38 and it also succeeded 🤦‍♂️.  So as sanity check, I ran the original again, this time setting it to do 2 rounds, and it failed with oom.  Finally I ran the patched version with 2 rounds and it succeeded again.  I think your patch works, but if you want me to test anything else or want the logs let me know.