PartialVolume/shredos.x86_64 #336: ShredOS vanilla version not booting

BrightDarknessXX, 2025-03-05:
I use Ventoy. I tried the .img and .iso vanilla. I see the booting ShredOS screen, and it's stuck there. Nomodeset works.

I want to help because I enjoy this project. What information do I need to gather for you?

PartialVolume, 2025-03-05:
The output of `lspci -k`, the make and model of the system or the motherboard model if it's a home build.

I don't know your level of experience with the Linux command line so if you have issues getting the lspci -k output onto your USB stick so you can post it here, let me know and I'll show you how.

BrightDarknessXX, 2025-03-15:
I use a windows 11 machine. My motherboard is the ASRock Z790 PG Lightning with BIOS version 16.02 (released 2025.01.16).
I don't know how to get lspci -k on the stick when booting the ISO with ventoy.

PartialVolume, 2025-03-15:
Once you have booted ShredOS and you can see the blue nwipe drive selection screen do the following:

With the ALT key held down press the F2 key which will display a black terminal with a prompt. At the prompt type the following

First we select the correct keyboard map, if you are using a US keyboard you won't need to do anything, if any other country you will probably need to use the loadkeys command with the two digit country code. For UK type, for other countries see the ShredOS readme.
```
loadkeys UK
```
Save the output of lspci -k to a file 
```
lspci -k > lspci.txt
```
Find the device name of your USB stick, usually /dev/sdb or maybe /dev/SDA if there are no hard drives attached to the SATA connectors.
```
fdisk -l | more
```
Create a directory that we will mount the USB flash drive on
```
mkdir /store
```
Mount the USB flash drive, replacing sdx with the device name of your USB flash drive found in step 1, it won't be sdx1
```
mount /dev/sdx1 /store
```
Copy the lspci.txt file to the USB flash drive
```
cp /lspci.txt /store/
```
Unmount the USB flash drive
```
cd /;umount store
```

That's it. Unplug the USB drive and plug it onto your Windows system to read the lspci.txt file.

Press the poweroff button to shutdown the PC or switch back to the nwipe screen with ALT F1

For the next version of ShredOS I will have ShredOS automatically copy the lspci.txt file to the USB stick when it boots up so you won't need to type all these commands.

BrightDarknessXX, 2025-03-15:
[lspcilog.txt](https://github.com/user-attachments/files/19263572/lspcilog.txt)

PartialVolume, 2025-03-15:
I only mention this because it worked with another user that had a Nvidia RTX 4080 Super graphics card. Try a bios reset and boot with ShredOS vanilla and see if that works.

BrightDarknessXX, 2025-03-15:
I tried to boot ShredOS Vanilla ISO with bios default setting by resetting my BIOS via the CLRMOS pins. No success. I tried normal and grub2 mode.

PartialVolume, 2025-03-15:
It looks like you also have onboard Intel graphics on the  motherboard. Are you plugging a monitor into both graphics ports. I'm just wondering if ShredOS vanilla is defaulting to the Intel onboard graphics instead of the Nvidia.

BrightDarknessXX, 2025-03-15:
I have 2 monitors which both go into my Nvidia RTX 4080 Super. One HDMI and one Display Port. I will check if BIOS enables integrated intel graphics.

BrightDarknessXX, 2025-03-15:
In BIOS, I have set ```Primary Graphics Adapter``` from ```Auto``` to ```PCIE1``` and ```IGPU Multi-Monitor``` from ```Auto``` to ```Disabled```. Tried ShredOS ISO Vanilla on normal and grub2. It did not boot. I have compared lspci -k output with the new settings and they are identical from the output before the changes.

PartialVolume, 2025-03-16:
This is all very reminiscent of this [post](https://github.com/PartialVolume/shredos.x86_64/issues/212#issuecomment-1922337894) and display port hardware.

You could see if unplugging the display port monitor so you only have the HDMI monitor attached allows ShredOS vanilla to boot.

On my Dell system I removed the Nvidia card (with display port) and attached a monitor to the embedded Intel graphics and it then worked ok having first set the bios to use onboard graphics as default.

There was some issue with the DRM drivers and display port connectivity, still it may be worth pulling that Nvidia card out and proving that's the problem.

PartialVolume, 2025-03-16:
I vaguely remember something about the display port monitor going to sleep and couldn't be woken. The LG monitors  amongst others suffered from this. It was something to do with power saving in the bios, a deep sleep mode that needed to be disabled. Maybe worth checking the bios for power saving options involving the display port monitor.

Searching the internet throws up many similar cases with multi monitor and card issues connected to display port and other distros like Ubuntu, mint and arch suffering from blank screen at boot.

BrightDarknessXX, 2025-03-16:
1. I unplugged my Display Port entirely and used HDMI only. Vanilla did not boot.
2. I changed my BIOS settings to use the integrated graphic from intel. I used the HDMI port for my integrated graphic. Vanilla booted up successfully.

My RTX 4080 Super was still inserted into the motherboard during testing.

BrightDarknessXX, 2025-03-16:
Is there any way to fix this to work on my NVidia Card?

PartialVolume, 2025-03-16:
To be perfectly honest, I've no idea. The problem seems to be a bug in the DRM driver code, maybe the amdgpu driver and while the whole DRM architecture is interesting https://docs.kernel.org/gpu/ I just don't have the time to dig deep into it, work out what the problem is and raise it as an issue on the kernel mailing lists. I'm pretty sure somebody must have already done this if you search the kernel mailing list archives.

It affects many different distros and seems to have been around for a few years at least. I did start reading the DRM developer documentation and with my very basic knowledge came to the conclusion it was a bug involving the connectors, a sub system that connects a frame buffer to a hardware device but like I say my knowledge of DRM is limited so I could be completely wrong. With each new kernel release I hope the bug gets fixed.

I'm hoping a developer in the DRM reads this and takes it upon themselves to identify the problem and resolve it as it does seem to impact systems with multiple graphics hardware but specifically Nvidia and display port.

Sorry I can't be of much use when it comes to this but hope it's fixed in the next kernel release.