PartialVolume/shredos.x86_64 #340: ERROR SG_IO bad/missing sense data and qla2xx instance issues, can not get to boot into main screen normally

errorx123, 2025-03-14:
Running lastest version of shredOs trying to secure erase a drive, working normally a few days ago, the suddenly I start getting qla2xxx instance errors and storage logs, never have these pops up before , then I got SG_IO bad missing data errors.

So instead of shredOs booting into the normal blue screen so I can select Which drive I want to wipe, I boot into this screen and I have to restart in order to get to the main screen.

I tried older versions, but get the same issue, different USB the same, I not know what happen I used to work just fine a few days before?

Images shows the messesages I got every time

https://imgur.com/a/eYm0r3c

PartialVolume, 2025-03-14:
[image]

From the above snapshot I can see you are running an older version v2023.08.2_25.0_x86-64_0.35. It also doesn't appear to be recognising any drives. When no drives are recognised nwipe will not display but will then show the archive_log.sh text shown at the bottom of the screen. I would suspect you have no SATA or IDE drives on your system and you just have a USB drive, or maybe you are using the exclude drives option and possibly your USB drive is not being recognised or is faulty? Can you post details using the latest version of ShredOS as lots of changes have been made to the nwipe_launcher since this version.

PartialVolume, 2025-03-14:
Something else that is odd about that snapshot above is every path/filename is preceeded with `^\\, also there is no text output from nwipe indicating that no drives were detected as if nwipe never get executed. 

If you could use that latest [ShredOS](https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2024.11_27_x86-64_0.38) in your tests. Thanks,

errorx123, 2025-03-15:
> [image]
> 
> From the above snapshot I can see you are running an older version v2023.08.2_25.0_x86-64_0.35 or earlier. It also doesn't appear to be recognising any drives. When no drives are recognised nwipe will not display but will then show the archive_log.sh text shown at the bottom of the screen. I would suspect you have no SATA or IDE drives on your system and you just have a USB drive, or maybe you are using the exclude drives option and possibly your USB drive is not being recognised or is faulty? Can you post details using the latest version of ShredOS as lots of changes have been made to the nwipe_launcher since this version.
> 
> 

Like I mention before in my post, I already test it with lastest version and get the same issue, I test 0.35 up to 0.38 , I used to 2 different usbs and the issue still persist, I have no satas in the PC is a PC that I use mainly to erase other drives, ATM of the screenshot I have no other drive connected.

errorx123, 2025-03-15:
> Something else that is odd about that snapshot above is every path/filename is preceeded with `^\\, also there is no text output from nwipe indicating that no drives were detected as if nwipe never [...]

Is weird indeed, but I already did some test with 0.38 and have the same results, any idea what is causing this, ATM I'm kinda busy but when I can  I will upload 0.38 screenshots.

PartialVolume, 2025-03-20:
>but this time I got a strange screen scrolling on first boot that says all pdf logs have been saved, press spacebar to restart, but when I press space bar nothing's happens I have to force shutdown?

That scrolling, with the message unknown command would suggest one of the keys is stuck or the keyboard maybe faulty.

Is this a laptop or PC? What's the make & model? What's the history of this system? Is it successfully running another O.S.?

errorx123, 2025-03-20:
> That scrolling, with the message unknown command would suggest one of the keys is stuck or the keyboard maybe faulty.
>
> Is this a laptop or PC? What's the make & model? What's the history of this [...]

Is an old laptop, I use only to secure erase hhd and sometimes I use it as a back pc, to do basic stuff, the model is a idea pad 14 Lenovo, it used too runs windows 10 , but nows days , I mainly use it to erase HDDs, and when I need I run tail or a unbuntu live usb, to do basic stuff, most of the time I use it without SSD connected, and lately only use shredos to boot and erase hhd.

Like I said before, it was working fine , but suddenly I was unable to boot normally to the main screen, I not know if there was a stuck key? the issue only happens once and I was not able to reproduce it again, did you check the other screen errors, you have an idea what could be happening, you think is the PC?

errorx123, 2025-03-22:
You now what's happening,  the full model is Lenovo 14iML05 ,  and in the comment about I describe the history behind it, any help?

PartialVolume, 2025-03-22:
What happens when you type  ALT F2 (hold ALT down while simultaneously pressing F2)

errorx123, 2025-03-23:
> What happens when you type  ALT F2 (hold ALT down while simultaneously pressing F2)

The first screen is the screen I got when I boot shredOs , then in that screen if I press Alt+F2 I got the scrolling down text.

Video and images 👇 

https://imgur.com/a/m4E21rp

PartialVolume, 2025-03-23:
It looks like a keyboard fault, I would open up the laptop disconnect the laptops keyboard and plug in a USB keyboard to prove it.

You're aware that nwipe won't be displayed unless a hard drive is detected (USB or built-in) and according to the images no drives are detected.

errorx123, 2025-03-24:
> It looks like a keyboard fault, I would open up the laptop disconnect the laptops keyboard and plug in a USB keyboard to prove it.
> 
> You're aware that nwipe won't be displayed unless a hard drive is detected (USB or built-in) and according to the images no drives are detected.

Done the test first video, is a normal boot
 Second video is holding alt+f2

https://imgur.com/a/ktsvKUc

https://imgur.com/a/eQB0SUf

What do you think about?

errorx123, 2025-03-25:
So far with the unplugged keyboard, It shows the results avobe, if I press F+2 I get the results of the second video ,

PartialVolume, 2025-03-25:
> So far with the unplugged keyboard, It shows the results avobe, if I press alt+f2 I get the results of the second video a line of sh-5.2# ?
> 
> This test was done with , the shredos USB connected, the external keyboard connected,and the laptop keyboard disconnected, no SSD or HDD connected.

Both of your most recent videos are as I would expect ShredOS/nwipe to behave, i.e they don't appear to show a problem.

I'm confused, is the scrolling text intermittent?

errorx123, 2025-03-25:
> Both of your most recent videos are as I would expect ShredOS/nwipe to behave, i.e they don't appear to show a problem.
>
> I'm confused, is the scrolling text intermittent?

I think I wouldn't explain well, what I was trying to say is that after, I unplug the keyboard of my laptop and connected a external key board I get a normal boot, and in this normal boot if I hold the alt+f2 key I get sh-5.2# on screen.

So that why I was asking if you see anything different, but it seems the issue could be the laptop keyboard , because when I plug in back to the laptop I get this screen even if I have a HDD connected to the laptop 👇 

https://imgur.com/a/kXI1vPW

PartialVolume, 2025-03-25:
Ah, ok. So it would seem it is a faulty keyboard. I was trying to workout which key might produce ^\ (control backslash) by trying various key combinations on my laptop but haven't figured it out yet. It's most odd as it seems to occur only at the beginning of an external command when a script is running.