PartialVolume/shredos.x86_64 #342: Customer/company information not being restored from ventoy USB drive.

PartialVolume, 2025-03-16:
### Discussed in https://github.com/PartialVolume/shredos.x86_64/discussions/341

<div type='discussions-op-text'>

<sup>Originally posted by **BrightDarknessXX** March 16, 2025</sup>
I want to add default customer or company information to the ShredOS ISO.
It doesn't seem to safe the customer information when I boot up ShredOS and add it via the options. It will be gone at the next start up.
What can I do?</div>

Confirmed as a bug. Working on a fix.

PartialVolume, 2025-03-18:
Added to project list https://github.com/users/PartialVolume/projects/1/views/1?pane=issue&itemId=102586896

PartialVolume, 2025-03-19:
@BrightDarknessXX I have completed a fix for this that works ok on my system. Basically I have extended the time from 5 to 10 seconds that there must be no further activity adding devices on the USB bus before an attempt is made to locate the USB device.

Are you in a position to test a new .img I've just built that contains this fix? If yes then I'll provide you with a link to download the .img

BrightDarknessXX, 2025-03-19:
I can test it. Could you provide me with the nomodeset version instead of vanilla?

PartialVolume, 2025-03-19:
Once you confirm whether this (v27.1) fixes the problem, I'll then commit the change. Thanks.

[shredos-2024.11_27.1_x86-64_0.38_20250319_nomodeset](https://drive.google.com/file/d/1t5tJgWj9_4inIMg2yZrcFBAle5-T3Leu/view?usp=sharing)

BrightDarknessXX, 2025-03-20:
Hey. I probably found additional complications. You have some extra options in Ventoy like Partition Style and Partition Configuration. For Style I use GPT and configuration I use NTFS. I don't wanna use exFAT here because I got some problems with another problem and I can't use FAT32 because I have bigger files than 4GB. Is there an easy way to get NTFS support?

BrightDarknessXX, 2025-03-20:
In the logs I could see that ShredOS did not find any FAT32/exFAT volume, which makes sense. Because there isn't any.

PartialVolume, 2025-03-20:
> Hey. I probably found additional complications. You have some extra options in Ventoy like Partition Style and Partition Configuration. For Style I use GPT and configuration I use NTFS. I don't wanna use exFAT here because I got some problems with another program and I can't use FAT32 because I have bigger files than 4GB. Is there an easy way to get NTFS support?

Yes you are correct, ShredOS is looking for a exFAT, Fat16, Fat32 formatted drive. I can add ntfs support. I'll add it to the project list.

BrightDarknessXX, 2025-03-29:
Ventoy seems to make it complicated?

PartialVolume, 2025-03-29:
When you boot ShredOS from Ventoy, type ALT F2 and then type fdisk -l. If you post a photo of the output. Thanks.

BrightDarknessXX, 2025-03-30:
[image]

PartialVolume, 2025-03-30:
Was there no partitions shown after the last line `Disk identifier ....` ? as in the example below.
```
Disk /dev/nvme0n1: 1.82 TiB, 2000398934016 bytes, 3907029168 sectors
Disk model: CT2000P3PSSD8                           
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 20462EB5-867F-3140-B376-8CE144C162EF

Device              Start        End    Sectors  Size Type
/dev/nvme0n1p1       4096     618495     614400  300M EFI System
/dev/nvme0n1p2     618496 3871647743 3871029248  1.8T Linux filesystem
/dev/nvme0n1p3 3871649792 3907028991   35379200 16.9G Linux swap
```

BrightDarknessXX, 2025-03-30:
My bad. I wrongly cut off the image.
[image]

PartialVolume, 2025-03-30:
I see the problem. Type is `Microsoft Basic Data`. I've no idea what that means, however that why it's not finding the drive. ShredOS looks for a type of fat16, Fat32 or exFAT.

I'll look into what `Microsoft Basic Data's` is supposed to represent and get back to you.

BrightDarknessXX, 2025-03-30:
I've read that Microsoft Basic Data marks the partition as one that contains data, unspecific to the filesystem.
In order to determine the filesystem, one has to read the boot sector or volume header.