PartialVolume/shredos.x86_64 #350: Can't Allocate Kernel on Dell Latitude 7400 models

DaveWeick, 2025-03-27:
Dear ShredOS team,

I can't boot the latest v0.38 | ShredOS .img x86_64bit for USB Vanilla from USB flash drive for the three Dell Latitude 7400 models I tried. After Grub and ShredOS have finished loading, I get message "error: can't allocate kernel" followed by "Failed to boot both default and fallback entries". Then, I'm back to the Grub choose menu, where I can choose to boot ShredOS again or switch to Command Line Mode, or edit parameters. 

Here's how I've isolated the problem so far:
- The previous v.0.37 boots fine on the Dell 7400. again using .img x86_64bit for USB Vanilla
- The latest v0.38 ( same USB stick ) boots fine on Dell Latitude 7390, Dell Optiplex 7090 micro, Dell Latitude 7420, Lenovo ThinkPad, Surface Go,
- Also tried a Ventoy USB with ShredOS .img x86_64bit for USB Vanilla. Same result on the Dell Latitude 7400
- BIOS versions on the 3 Dell Latitude 7400 were 1.21, 1.30 and 1.32.  There is a more recent 1.33 BIOS version ( not tried ). Processor is Intel Core i7-8665U @1.90 GHz
- BIOS is in UEFI mode, SATA/Storage operation is AHCI and Secure Boot is disabled. I used Rufus 4.6 to create the USB.

ShredOS has been a great tool for my business and I'd like to move up to the latest version if it weren't for this issue. Please advise, including how I might collect further diagnostic information.

Firminator, 2025-03-28:
maybe try https://github.com/PartialVolume/shredos.x86_64/issues/308

krause-wclc, 2025-06-05:
We're having the same issue. :(

PartialVolume, 2025-08-20:
Thanks for the update, good to know.

JoeArcher007, 2025-09-03:
I was running into the same issue on a Dell XPS 13 (I think 94XX or something along those lines).  tried various ways of getting it to start but found it would do the same. Load all the way to 100% and then just say failed to allocate.

I then turned off secure boot and turned on legacy CSM mode, and ensured to boot it from CSM and that worked. Was able to get it up and running.

I'd love to know why it wouldn't allocate properly from UEFI mode, but CSM may do the trick for you if you just want to shred the disk locally.

closed 2026-02-14 by PartialVolume