PartialVolume/shredos.x86_64 #354: Error when wiping disk, stop midway 0 error shows in the report

errorx123, 2025-04-20:
I was secure erase a disk, when I notice the screen shows the image below, in the error section shows zero errors, this is a new laptop HP pavilion x360 ek0010la.

Other thing I notice was, this line quote :
Cp :cannot stat' /nwipe_report *pdf: no such file or directory.

Any idea what is happening?

Error image bellow 

https://imgur.com/a/0mNIjlz

kduvekot, 2025-04-22:
@PartialVolume looks like a duplicate of #332

errorx123, 2025-04-28:
> @PartialVolume looks like a duplicate of #332 

Is there any solution for this issue? Or why cause i search in that threat and I not exactly understand the cause of the issue!

kduvekot, 2025-04-28:
@errorx123 plug in a network cable to a network that provides DHCP ... Or download the previous version and use that.

PartialVolume, 2026-02-14:
Closing as most probably fixed in https://github.com/PartialVolume/shredos.x86_64/releases/tag/v2025.11_28_x86-64_0.40

closed 2026-02-14 by PartialVolume