PartialVolume/shredos.x86_64 #355: Shredos cannot allocate kernel

theexeture, 2025-04-22:
having an issue to boot into shredos on latest version. I have tried ventoy as well but all bring me the same issue where it loads to 100 percent then says cannot allocate kernel and brings me back to grub menu.

this is my laptop model. no changes made but laptop is slightly old.

https://www.acer.com/us-en/laptops/aspire/aspire-5-intel/pdp/NX.A2AAA.001#pdpSpecs

krause-wclc, 2025-06-05:
Any update? We're having this issue as well. I can update with the specs here.

JoeArcher007, 2025-09-03:
I was running into the same issue on a Dell XPS 13 (I think 94XX or something along those lines).  tried various ways of getting it to start but found it would do the same. Load all the way to 100% and then just say failed to allocate.

I then turned off secure boot and turned on legacy CSM mode, and ensured to boot it from CSM and that worked. Was able to get it up and running.

I'd love to know why it wouldn't allocate properly from UEFI mode, but CSM may do the trick for you if you just want to shred the disk locally.

closed 2026-02-14 by PartialVolume