PartialVolume/shredos.x86_64 #358: Scrambled screen on start

RandomEngy, 2025-05-27:
Used Rufus to create an image from v2024.11_27_x86-64_0.38 , with the USB Vanilla.

This is what happens when I UEFI boot from the USB drive:
[image]

This is what happens when I choose the "USB" option:
[image]

I also see the scrambled screen when I use Ventoy. The Ventoy UI shows correctly, but it shows scrambled when attempting to launch the ShredOS image.

Secure boot is disabled.

ASRock X570 Steel Legend

PartialVolume, 2025-06-19:
Thanks for letting us know 👍

PartialVolume, 2025-10-23:
Closing as resolved.