PartialVolume/shredos.x86_64 #368: Smartctl is unable to provide smart data for /dev/sdb

Rainywaters1, 2025-07-25:
Hello, 

I am constantly getting this error message when trying to wipe. Please see the log file attached and video. Is there anything I can change in the BIOS to fix this?

[nwipe_log_20250724-101233.txt](https://github.com/user-attachments/files/21434507/nwipe_log_20250724-101233.txt)

https://github.com/user-attachments/assets/10ba37a4-a859-4392-af18-b063643bd6d5

PartialVolume, 2025-07-25:
You have two different problems.

1. Unable to obtain smart data for /dev/sdb
2. Video shows I/O error.

1. According to the logs you are wiping a Buffalo SSD attached via the USB interface. No amount of changes to the bios will fix this problem as it's most likely the chipset  being used by the Buffalo SSD does not support ATA pass through and so the commands used to interrogate the device don't work. Possibly not much you can do about that other than remove the SSD and connect it to a M.2 connector on a motherboard assuming the SSD has the appropriate interface.

2. The I/O occurs because a read/write error was detected. Assuming you didn't unplug the USB device during a wipe which would cause an I/O error and there is nothing flaky with the USB connector then it's most likely you SSD has some errors on it.

avefrandsen, 2025-07-28:
ShredOS is very small, you can put it on just about any flashdrive still floating around in the world. If you're running it a lot, you probably want a USB3 thumbdrive so that it boots very fast instead of pretty fast. If you're wiping large batches of drives and generating PDFs you will want *several gigabytes* to store them in.

I have to ask, why are you trying to put ShredOS on an SSD, boot to it, and then wipe that same SSD?

Rainywaters1, 2025-07-28:
> ShredOS is very small, you can put it on just about any flashdrive still floating around in the world. If you're running it a lot, you probably want a USB3 thumbdrive so that it boots very fast [...]

I have ShredOS on an external SSD which is the Buffalo SSD, then I have an external Unitek adapter model S1232A that allows ATA passthrough to wipe a hardrive. Please see the image below

[image]

PartialVolume, 2025-07-28:
It's likely an issue with smartctl due to the Unitek S1232A having two  interfaces, M2 and SATA. I had this issue with another multi interface Unitek USB adapter and spoke to the maintainer of smartctl. After providing technical information about the chipset in use he was able to modify smartctl so it correctly handled retrieval of smart data from both interfaces.

Can you boot ShredOS, makesure the Unitek adapter is plugged in and the run `lsusb` and post the output here.  It should mention the chipset in use for the Unitek.

Thanks.

PartialVolume, 2025-07-28:
I've just checked the latest version of smartmontools, which is 7.5, released in May, the Unitek commit I linked to was committed in April so the fix mentioned should be in smartmontools 7.5. The latest ShredOS uses Smartmontools 7.3 so looks like it's time for me to release another update that incorporates 7.5.

Rainywaters1, 2025-07-29:
> So it's the same device ID as the device I have so more than likely it's the same chipset.
> 
> I guess your options are:
> 
> * Wait until I release an updated ShredOS, I can't give a timescale on that as I have other commitments but hopefully soon.
> * Try one of the other Linux distros that run the latest nwipe with the latest smartmontools so that you can prove your Unitek USB adapter now provides the smart data.
> 
> https://github.com/martijnvanbrummelen/nwipe?tab=readme-ov-file#which-linux-distro-uses-the-latest-nwipe

Ok, thanks for the help! I purchased this device because it was on the list of USB adapters that contain a SATA controller chip that supports ATA pass through for secure erase/HPA/DCO/smart data. Can you recommend another device to use that will allow NVMe drives to be wiped in addition to the harddrives? This was the only one I could find.

Rainywaters1, 2025-07-30:
> > I have ShredOS on an external SSD which is the Buffalo SSD, then I have an external Unitek adapter model S1232A that allows ATA passthrough to wipe a hardrive. the hard drive ST500DM0 02-1BD142, 500 GB is what I am trying to wipe. Is it possible the combination of the system, adapter, and ShredOS is leading to I/O errors or a lack of drive detection. Please see the image below
> 
> I just want to specifically draw attention to the detail that in your original video, NWipe is trying to wipe a Toshiba DT01ACA050, not a Seagate ST500DM002. One would presume that the DT01ACA050 is in the Buffalo external drive, and if true you aren't trying to wipe that, but also the DT01ACA050 is a 2.5" spinning disk hard drive, not an SSD. That won't cause any problems in all this, but my experience with Buffalo is they make bad products and lie about them.
> 
> Best case this doesn't matter to you, or you want to return the drive. Worst case, including a low performing enclosure with a low performing hard drive in with the ST500DM002 may cause the wipe group to take several times longer. Once the Seagate is visible, don't select the Toshiba.
> 
> > Can you recommend another device to use that will allow NVMe drives to be wiped in addition to the harddrives?
> 
> SATA and NVMe are completely different technologies that require unrelated hardware, this makes it hard to find docks/adapters that support both and a lot of the time they have to make sacrifices to keep the price down. It's trivial to make an adapter that supports traditional SATA and M.2 SATA, since they are the same protocols on the same sets of wires in differently shaped ports. Read the fine print as sometimes these only support 1 drive, regardless of how many can be connected at the same time. That would actually bypass this specific issue, but since the fix is on it's way it will be less capable later.
> 
> I feel you're best off finding a separate device for NVMe, such as an NVMe M.2 external enclosure. There's quite a lot of these that are tool-less and use type-C cables. If your host computer has a Thunderbolt port, then a Thunderbolt enclosure will literally connect the NVMe drive as a PCIe device, same as if it was in an M.2 port on the motherboard.

Thank you for this information! I purchased a USB3 and will try that today instead of using the Buffalo SSD to boot ShredOS. Before I started using the Buffalo SSD I had a USB2 and it worked great. I just thought by switching to the SSD it would run faster and easier, but I've had nothing but issues with it. I hope that works better. I am not extremely tech savvy, so I appreciate all the insight!

Rainywaters1, 2025-07-31:
Update, I used a USB3 and still received the same results. I will wait on the fix to resolve the issue.