PartialVolume/shredos.x86_64 #373: nwipe v0.39

jasonbot1179, 2025-09-17:
Greetings,

Will ShredOS be updated for v0.39 of nwipe ?

https://github.com/martijnvanbrummelen/nwipe/releases/tag/v0.39

Thank you for this operating system!

DaanSelen, 2025-10-16:
> > [@PartialVolume](https://github.com/PartialVolume) is this not just bumping the version and checksum, I can make a PR for that?
> 
> Yes, please go ahead with that. Thanks.
> 
> You may not need this info but thought I'd include this information from my notes in regards to the hash.
> 
> ** nwipe.hash ** This file needs to contain the following line with a sha1 hash taken from the nwipe tar.gz file that is obtained from https://github.com/martijnvanbrummelen/nwipe/releases Once you have downloaded the appropriate version of nwipe, run sha1sum nwipe.v0.xx.tar.gz. copy the hash into the file nwipe.hash. The format of this line is important, if you don't include the preceding sha1 text the compile will fail with a 'unable to find hash for nwipe.v0.xx.tar.gz'

I managed to get it to work. I succesfully booted off the .img file I made myself and MY GOODNESS, compilation takes LONG!
Making a PR within 24 hours.

DaanSelen, 2025-10-21:
I am checking this all out, but I am unsure if the PDF generation is working along with nwipe's new version.